ChristianTremblay/BAC0 #19: Provide more infos on the docs

noetius, 2016-04-20:
I've noticed that in the docs there aren't explanations about how to:

- Define a controller providing an IP address rather than an MSTP like in the example. Trying to do that was unsuccessful cause device appears disconnected to BAC0. Probably another question could be: BAC0 is able to connect to a controller with an IP address?

- Read the value of a specific point, what does it mean point name? The object property name? What is the correct notation within square brackets in order to read or to write an object's property?

Is it possible to provide these infos?

ChristianTremblay, 2016-04-21:
Regarding IP
I don't have access to many IP devices... the ones I've been able to connect to were quite easily accessed. You need to know the IP address and the device ID.
Code would look like : 
```
import BAC0
bacnet = BAC0.connect()

controller = BAC0.device('ip', deviceId, bacnet)
```

As a starting point, to verify if your computer sees some devices, you can use (once `bacnet`is defined) : 
```
bacnet.whois()
bacnet.devices
```
it'll list every devices on your network

ChristianTremblay, 2016-04-21:
Regarding Point
I may suffer from semantic approximation. It's a term we always use to define a variable in the controller. In fact, it would be the object name.

Once your device is created, use : 
```
controller.points
```

It will list every points that have been included in the device (analogInput, analogOutput, analogValue, multistateInput, multistateValue, binaryInput, binaryOutput, binaryValue) Note that schedules or trends have not yet been included in the device.

That said, `controller.points`will shoiw you a list then it's easy to choose from the list the point you need to interact with. 

Example : 
```
controller['this object with spaces in its name']
```
(you pass the name as a string in the brackets)

noetius, 2016-04-21:
Thanks for your reply,
well the problem maybe is that i'm not able to see controllers around me too, although they're on the same subnet where I'm running python scripts with BAC0, a simple network bridged lubuntu vm.
Since BAC0 is built upon bacpypes, i've tried to read some controllers objects properties first with bacpypes on 47808 udp port with a successful outcome (bacpypes/samples/ReadProperty.py), then I tried with BAC0, taking also advantage of the ReadWriteScript package but i get only NoResponseFromController exceptions.

For the moment I can't understand where really is the problem. 

BAC0 is a promising project, and I would want take some time in order to testing its potentials, but it's really hard if I can't establish any kind of connection.

ChristianTremblay, 2016-04-21:
You will need Wireshark to see what's going on your network.
Also, you may need to define your IP/subnet manually when defining the bacnet network instead of using what's found automatically by the script.

Let's say you have a really simple subnet, you could try : 
```
import BAC0
bacnet = BAC0.connect(ip='192.168.1.10/24')
# This will use your network card configured for IP 192.168.1.10
# /24 means using mask : 255.255.255.0
```

It's possible that actually, BAC0 tries to find controller on your wireless card (for example) but controllers are on the subnet of your wired card.

The `connect()` function finds the adapter that is connected to internet

noetius, 2016-04-22:
Thanks for your time, I was able to find devices after some adjustaments.
Do you think it might be worthwhile to expose another attribute in BAC0.connect method to allow udp connection over the local network to another port other than 47808?

ChristianTremblay, 2016-04-22:
Did you try ?
```
import BAC0
bacnet = BAC0.connect(ip='192.168.1.10:47809/24')
```

ChristianTremblay, 2016-04-22:
I would really appreciate if you could propose some improvements to documentation based on your experience. Doc is a big challenge as I am so involve, I may forget some details that are really important to those who begin working with the tool.

This particular file : 
https://github.com/ChristianTremblay/BAC0/blob/master/doc/source/connect.rst

noetius, 2016-04-26:
I've tried to add two more examples to connect.rst... despite my inexperience with rst templates.

ChristianTremblay, 2016-04-26:
Great ! You can make a Pull Request and I'll merge your new examples... I'll probably tweak the subtitles but nothing big. Or you could, just before the PR...

I would write
```
Examples
-------------
Connecting to MSTP device
======================
bla

Connecting to IP device
======================
bla

Connecting to IP device on different port
===============================
bla
```

I would also add some "generic" reference to the address format... Something like : 
`ip:port/mask` ex. `192.168.10.10:47809/24`

What do you think ?

noetius, 2016-04-27:
Perfect, it seems a good idea.

ChristianTremblay, 2016-05-05:
Do you want to do the PR or do you prefer I just modify the file ?

noetius, 2016-05-05:
It would be fine if you did it in my place... i've tried rst editor but it was a totally mess

closed 2017-03-27 by ChristianTremblay

Itja, 2018-01-26:
Note for anyone hitting this on google:
The syntax for a custom port is described [here](https://github.com/JoelBender/bacpypes/blob/856cad7a7ba4252a07d362f4e98cb395e7b2bf53/py34/bacpypes/pdu.py#L31) and therefore is
```
import BAC0
bacnet = BAC0.connect(ip='192.168.1.10/24:47809')
```

So it is not first IP, then port, then mask like shown in [this comment](https://github.com/ChristianTremblay/BAC0/issues/19#issuecomment-213515569), but it's **IP/Mask:Port**.