ChristianTremblay/BAC0 #21: TypeError: argument of type 'int' is not iterable - read_mixin.py

noetius, 2016-04-22:
After establishing connection successful to a device, I've simply tried to read a point value, and then this error appeared:

<pre>
<code>
File "/BAC0/core/devices/mixins/read_mixin.py", line 173, in retrieve_type

if point_type_key in point_type:

TypeError: argument of type 'int' is not iterable
</code>
</pre>

After giving a look to code, I've simple edited file with this quick-n-dirt fix:

<pre>
<code>
if point_type_key in str(point_type):
</code>
</pre>

Probably not the best approach, but for the moment it works.

ChristianTremblay, 2016-04-22:
In core/function/discoverPoints.py line 49, there's a "manual"exclusion list of point actually not supported by BAC0.

In your case, I'm curious to know which type of point failed.

Or is this only that the name of the point that was an integer ?

Can you provide the object_list ? (maybe the type of controller and manufacturer) it could help

noetius, 2016-04-26:
Sure.

_**Controller**_: DAC-633E Application Controller
_**Manufacturer**_: Delta Controls Inc.

_**obj_list**_:

**point_type_key, point_type**
analog, analogInput
analog, analogInput
analog, analogInput
analog, analogInput
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, analogValue
analog, binaryInput
analog, binaryInput
analog, binaryInput
analog, binaryInput
analog, binaryOutput
analog, binaryOutput
analog, binaryOutput
analog, binaryOutput
analog, binaryOutput
analog, binaryOutput
analog, binaryOutput
analog, binaryOutput
analog, device
analog, eventEnrollment
analog, file
analog, multiStateInput
analog, multiStateInput
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, notificationClass
analog, program
analog, schedule
analog, schedule
analog, 142 **(int that raised error)**

ChristianTremblay, 2016-04-26:
Haaa! A proprietary property. I'll have to think about a way to allow the usage of those kind of properties. But actually, I will find a way to exclude them from the list.

Including them will mean "probably" a distinct class defining how this property works...

Is this property absolutely necessary in your tests ?

noetius, 2016-04-26:
Not for the moment. But i think it would useful don't exclude it, because I don't understand what concerns this object type.

ChristianTremblay, 2016-04-26:
hummm we'll have to think about it.... and we'll have to treat 142 as a string... not a int for getter (controller['142']) to work

ChristianTremblay, 2016-05-05:
As those vendor object are not standard, I will not be able to keep them accessible via the square bracket getter.
BUT, I could implement a function in the device allowing to read "non standard object" 

Maybe something like : 
```
controller.prop('142','presentValue')
```

prop = proprietary

The user must know details about the property to use the function...

ChristianTremblay, 2018-03-22:
@noetius  Proprietary object can now be created. See develop branch where you will see one example for a JCI proprietary value for the TEC3000 thermostat.
See : BAC0\core\proprietary_objects

closed 2018-09-29 by ChristianTremblay