ChristianTremblay/BAC0 #32: Question:

Grogglex, 2017-03-27:
In the [documentation on using bokeh](http://bac0.readthedocs.io/en/latest/trending.html#add-plots-to-bokeh-document) to plot points I was wondering where you're getting the 'nvo' value from. I understand that AI2 = Analog Input 2. 

```
# This script will add four plots to the server
lst = ['nvoAI1', 'nvoAI2', 'nvoDO1', 'nvoTempRdC']
fx.chart(lst, title = 'First Floor temperature')
```

Grogglex, 2017-03-27:
I understand now that the points listed are actually points from the device, I just didn't see that kind of notation before. Sorry about that.

closed 2017-03-27 by Grogglex

ChristianTremblay, 2017-03-27:
The notation is due to the kind of controller I was using to test the software when I was doing the doc. Those were legacy Johnson Controls FX Line devices. And they were compatible with N2, Lonworks and BACnet. But to comply with the Lonworks protocole, variables names were using the Lonworks naming scheme. nvoSomething nviElse, nciOther... But yes as you pointed, it's only variable names.