ChristianTremblay/BAC0 #34: sigterm will fail if core.run() is called in a thread (bacpypes issue #119)

ChristianTremblay, 2017-03-31:
See e1dc495 for fix in develop branch

MichelleNing, 2017-04-07:
I use BAC0 -- 0.99.82, bacpypes -- 0.15.2.
After I import BAC0, it reports:
    xlwings not installed. If using Windows or OSX, install to get more features.
When I try to create bacnet connection, bacnet = BAC0.connect(ip='192.168.0.2'), it reports:
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\ProgramData\Anaconda3\lib\threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "C:\ProgramData\Anaconda3\lib\site-packages\bacpypes\core.py", line 97, in run
    signal.signal(signal.SIGTERM, sigterm)
  File "C:\ProgramData\Anaconda3\lib\signal.py", line 47, in signal
    handler = _signal.signal(_enum_to_int(signalnum), _enum_to_int(handler))
ValueError: signal only works in main thread

Is it the same issue why you create this bacpypes issue? 
Is there any way I can successfully use BAC0.connect()?
Thank you~~~

ChristianTremblay, 2017-04-08:
Will be included in next release

closed 2017-04-08 by ChristianTremblay

MichelleNing, 2017-04-10:
Hello Christian,

Many thanks for your reply. I have a question, I cannot find my BACnet
device. The device is shown online, and I can use other tool to find it.
However, when I use BAC0.device, it shows as:

>>> bacnet = BAC0.connect(ip='192.168.0.2')

xlwings not installed. If using Windows or OSX, install to get more features.
Starting app...
App started
Starting Bokeh Serve
Click here to open Live Trending Web
Pagehttp://localhost:5006/?bokeh-session-id=JgCiWr5OWZesb0FPfI5YiBEwe1Ozd4NWZeBGTYaQmlAA
Failed running bokeh serve
Bokeh server already running

>>> bacnet

Bacnet Network using ip 192.168.0.2 with device id 3056654

>>> fx = BAC0.device('3:96', 96, bacnet)

Changing device state to <class 'BAC0.core.devices.Device.DeviceDisconnected'>
Provide dbname to connect to device offline
Ex. controller.connect(db = 'backup')

Could you give me some suggestion?

Thank you for your attention on my issue.

Best regards,

Michelle

ChristianTremblay, 2017-04-10:
you can try : 

```
bacnet.whois()
bacnet.devices
```
it will tell you who is on the network

ChristianTremblay, 2017-04-10:
If this doesn't work, please open another issue. Just to be sure we don't mix things.
thanks !

MichelleNing, 2017-04-10:
Hello Christian,

Thank you for your suggestion.
I can find device information using bacnet.whois() and bacnet.devices.
However, when I use BAC0.device, it still replys me device offline.
I list what I got as follows:
>>>  bacnet.whois()

defaultdict(int, {('192.168.0.16', 1000): 1, ('3:96', 96): 1})

>>> bacnet.devices

Manufacturer Address Device ID
Name
BACnet Router FieldServer Technologies 192.168.0.16 1000
SER835x Schneider Electric 4:36 96

>>> fx = BAC0.device('3:96', 96, bacnet)

Changing device state to <class 'BAC0.core.devices.Device.DeviceDisconnected'>
Provide dbname to connect to device offline
Ex. controller.connect(db = 'backup')

As shown in two command, the device SER835x got two different Address.
For bacnet.whois, its address is 3:96. For bacnet.devices, its address
is 4:96

It is right? Thank you so much!

Best regards,

Michelle