ChristianTremblay/BAC0 #35: Question: Read the Value of a Single Point

Grogglex, 2017-03-31:
So I'm finding that connecting to a device with a lot of points is taking a very long time. One of our devices has 1000 points and it takes somewhere around 10m to connect to that device. I was just wondering if BAC0 has support for something like `bacrpm`, to read the present value of a single point without creating a controller object.

Additionally, people who use the library, do you tend to keep multiple devices connected at the same time in order to speed up access? 

Thanks in advance!

ChristianTremblay, 2017-03-31:
By default, BAC0 will use read property multiple, unless segmentation is not supported by the device. !0 minutes seems a really long time and I would suspect that RPM is not used... Hard to tell from here though.

In any time, you can use pure bacpypes syntax to reach a point.
```
import BAC0
bacnet = BAC0.connect()
bacnet.read('2:6 analogInput 1106 presentValue')
bacnet.readMultiple('2:6 analogInput 1106 presentValue description')
```

I've developed this software with the idea of temporary communication with devices (as I make tests on the sequence or while commissionning it).
So there may be some improvements needed to keep devices connected for long period of time.
Probably some memory management as all the histories are kept in-memory...

I'm open to suggestions

ChristianTremblay, 2017-03-31:
To find the information about a point, you can first define the device.
Then call the point list...
Identify the variable of interest and check the "properties"

```
fx = BAC0.device('2:6',55006,bacnet)
```

> Changing device state to <class 'BAC0.core.devices.Device.DeviceDisconnected'>
> Changing device state to <class 'BAC0.core.devices.Device.RPMDeviceConnected'>
> Found PCV-2-006... building points list
> Ready!
> Polling started, every values read each 10 seconds

```
fx.points
```

> [DA-VP : 9.48 pascals,
>  ZN-T : 23.27 degreesCelsius,
>  DA-T : 533.64 degreesCelsius,
>  DPR-O : 100.00 percent,
>  OA-PERCENT : 100.00 percent,

```
fx['ZN-T'].properties
```

> {'device': PCV-2-006 / Connected, 'type': 'analogInput', 'simulated': (False, 0), 'address': '1106', 'name': 'ZN-T', 'units_state': 'degreesCelsius', 'description': 'Zone Temperature', 'overridden': (False, 0)}

Grogglex, 2017-03-31:
Thanks! I might actually go with the [basic scripts](http://bac0.readthedocs.io/en/latest/BAC0.scripts.html#module-BAC0.scripts.ReadWriteScript) for our tasks. Our issue is that the points with 1000+ don't seem to work with `BAC0.device('2:6',55006,bacnet)`.

closed 2017-03-31 by Grogglex

ChristianTremblay, 2017-03-31:
I suspect it doesn't support segmentation. I would not be surprised it work vith `BAC0.device('2:6', 55006, bacnet, segmentation_supported=False)`
It would explain why it is so slow

ChristianTremblay, 2017-03-31:
As you can see in the __init__.py, when you call 
```
bacnet = BAC0.connect()
```
You are actually defining a ReadWriteScript
```
from .scripts.ReadWriteScript import ReadWriteScript as connect
```