ChristianTremblay/BAC0 #38: BAC0.device cannot find BACnet device

MichelleNing, 2017-04-11:
I used the fix of issue 34 on Window.
I list what I got as follows:
>>  bacnet.whois()
defaultdict(int, {('192.168.0.16', 1000): 1, ('3:96', 96): 1})
>> bacnet.devices
Manufacturer Address Device ID
Name
BACnet Router FieldServer Technologies 192.168.0.16 1000
SER835x Schneider Electric 4:36 96
>> fx = BAC0.device('3:96', 96, bacnet)
Changing device state to <class 'BAC0.core.devices.Device.DeviceDisconnected'>
Provide dbname to connect to device offline
Ex. controller.connect(db = 'backup')

As shown in two command, the device SER835x got two different Address.
For bacnet.whois, its address is 3:96. For bacnet.devices, its address
is 4:96
Sometimes, it shows the correct address, like
>> bacnet.devices
Manufacturer Address Device ID
Name
BACnet Router FieldServer Technologies 192.168.0.16 1000
SER835x Schneider Electric 3:96 96

However, no matter what bacnet.devices replied, I cannot find BACnet device by using BAC0.device('3:96', 96, bacnet)

ChristianTremblay, 2017-04-12:
This is weird. Have you tried `BAC0.device("4:96",96,bacnet)` ?

Also, you can always use bacpypes syntax to help...
using for example, `bacnet.read('3:96 analogInput 1 presentValue')` or `bacnet.read('3:96 device 96 objectName)` This can help browsing the network before your work faster than defining a device.

You will also need to be sure that there is no network configuration error. The SER device seems to be an MSTP device behind a BACnet router. Be sure the router it is well defined.

If simple `bacnet.read` commands works, maybe there is a bug with the device creation for this device. We will need to dig further.

ChristianTremblay, 2017-04-12:
Just had a flash... I remember you said that you defined your connection using : 
`bacnet = BAC0.connect(ip='192.168.0.2')`

This will result in using a broadcast address of 255.255.255.255 which in certain case can lead to problem.

Please, test using
`bacnet = BAC0.connect(ip='192.168.0.2/24')`
This will force using the broadcast address 192.168.0.255
(I here presume that your mask is 255.255.255.0)

MichelleNing, 2017-04-12:
My network mask is 255.255.255.0
I tried
>>> bacnet = BAC0.connect(ip='192.168.0.2/24')
>>> bacnet.whois()

defaultdict(int, {('192.168.0.16', 1000): 1, ('3:49', 81049): 1})

>>>bacnet.devices

BACnet Router FieldServer Technologies 192.168.0.16 1000
SER8150xxxxx-49 Schneider Electric 3:49 81049

>>> bacnet.read('3:49 device 81049 objectName')

'SER8150xxxxx-49'

>>> bacnet.read('3:49 analogValue 100 presentValue')

27.44444465637207

All above outputs are correct, however, when use BAC0.device, I still
cannot find device:

>>> controller = BAC0.device('3:49', 81049, bacnet)

Changing device state to <class 'BAC0.core.devices.Device.DeviceDisconnected'>
Provide dbname to connect to device offline
Ex. controller.connect(db = 'backup')

Thank you.

ChristianTremblay, 2017-04-12:
There is at least some progress....

Can you poll : 
`bacnet.read('3:49 device 81049 objectList')

it is the variable I try to read when building a device

MichelleNing, 2017-04-12:
When I poll bacnet.read('3:49 device 81049 objectList'), I got the following information:
>>> bacnet.read('3:49 device 81049 objectList')
NoResponseFromController                  Traceback (most recent call last)
<ipython-input-8-63bffff2f55e> in <module>()
      7 #bacnet.read('3:49 analogValue 100 presentValue')
      8 #controller = BAC0.device('3:49', 81049, bacnet)
----> 9 bacnet.read('3:49 device 81049 objectList')

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\io\Read.py in read(self, args, arr_index)
    133         # do something for error/reject/abort
    134         if iocb.ioError:
--> 135             raise NoResponseFromController()
    136 
    137             # Share response with Queue

NoResponseFromController:

ChristianTremblay, 2017-04-12:
This is why you cannot create a device from this controller. BAC0 will use the objectList provided by the device to populate all the variables...
I can't tell if it's because its using an old standard or if it's just a wrong implementation of BACnet...

In your particular case, you will have to interact with the device using bacnet.read 

I will poke aound and see how I could handle this.

ChristianTremblay, 2017-04-13:
@MichelleNing I've verified and the objectList should be implemented for a device to pass conformance. But it seems that it is not the case for your device.

It could be possible to modify the BAC0.device class to allow a object_list dict to be passed at the creation of the device. 
You would need to build something manually : 
```
my_object_list = [ ('device', 123),
                             (analogInput', 1),
                             (analogValue', 1),
                             ('binaryInput', 2)
                             #etc... 
                              ]
```
We could then be able to define an object using "only" the objects we send to the device

```
my_device = BAC0.device('3:4, 5004, bacnet, object_list = my_object_list)
```

MichelleNing, 2017-04-18:
I modified the BAC0.device class to allow a object_list
```
my_object_list = [ ('3:66', 81066),
                  ('analogInput', 1),
                  ('analogOutput', 1),
                  ('analogValue', 1),
                  ('binaryInput', 2),
                  ('binaryOutput', 2),
                  ('binaryValue', 2),
                  ('multiStateInput', 3),
                  ('multiStateOutput', 3),
                  ('multiStateValue', 3)]
```
`controller = BAC0.device('3:66', 81066, bacnet, my_object_list)`

I got:
Changing device state to <class 'BAC0.core.devices.Device.DeviceDisconnected'>
Provide dbname to connect to device offline
Ex. controller.connect(db = 'backup')

I might be use other functions instead of BAC0.device.

Thank you

ChristianTremblay, 2017-04-19:
Please have a look to the develop branch. I've modified the code to be able to provide a custom object list.

Even if in your case, you need this because of a bad implementation , I found this feature interesting if I would need to define an object with only the points I needed.

Will need tests but I'm pretty sure it works ok.

MichelleNing, 2017-04-19:
I cope BAC0 folder which is git cloned from develop branch. 
When I run BAC0.device, it replied: ERROR:BAC0.core.devices.Device:Segmentation not supported

May I know where I can get BAC0 version 0.99.72? Because after I pip install BAC0==0.99.72, it didn't work. I tried to find tag for 0.99.72, but I didn't find it. My friends told me, 0.99.72 worked very well.

Thank you so much!

ChristianTremblay, 2017-04-19:
Thanks for reporting. This should not be an error but a warning.
One thing I know when doing multiple test with a lot of `python setup.py install` is that sometimes,, things get screwed. 
I suggest you `pip uninstall BAC0` (repeat) until there is no version left on your system. Then retry to install.

If you would be kind enough to do one more test, it would be very appreciated. I don't have access to a lot of different devices so when I find some weird ones, I like to be able to make them work.

Please try (with the latest develop code)
```
# your own version...
my_object_list = [ ('3:66', 81066),
                  ('analogInput', 1),
                  ('analogOutput', 1),
                  ('analogValue', 1),
                  ('binaryInput', 2),
                  ('binaryOutput', 2),
                  ('binaryValue', 2),
                  ('multiStateInput', 3),
                  ('multiStateOutput', 3),
                  ('multiStateValue', 3)]

controller = BAC0.device('3:66', 81066, bacnet, object_list=my_object_list, segmentation_supported=False)
```

Regarding the version 0.99.72 I will try to look at it later today... I must admit that BAC0 is one of my really first project and I confess a lot of git error since the beginning. Sorry about that.

MichelleNing, 2017-04-19:
When I use Jupyter QtConsole and Jupyter Notebook, I got:
`my_object_list = [ ('3:66', 81066),
                  ('analogInput', 1),
                  ('analogOutput', 1),
                  ('analogValue', 1),
                  ('binaryInput', 2),
                  ('binaryOutput', 2),
                  ('binaryValue', 2),
                  ('multiStateInput', 3),
                  ('multiStateOutput', 3),
                  ('multiStateValue', 3)]

controller = BAC0.device('3:66', 81066, bacnet, object_list=my_object_list, segmentation_supported=False)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-8-dfd2c52d03a0> in <module>()
----> 1 controller = BAC0.device('3:66', 81066, bacnet, object_list=my_object_list, segmentation_supported=False)

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in __init__(self, address, device_id, network, poll, from_backup, segmentation_supported, object_list)
    149                 raise FileNotFoundError("Can't find %s on drive" % filename)
    150         else:
--> 151             self.new_state(DeviceDisconnected)
    152 
    153 

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in new_state(self, newstate)
    160         self._log.info('Changing device state to {}'.format(newstate))
    161         self.__class__ = newstate
--> 162         self._init_state()
    163 
    164 

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in _init_state(self)
    637     """
    638     def _init_state(self):
--> 639         self.connect()
    640 
    641 

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in connect(self, db)
    665                     self.new_state(RPMDeviceConnected)
    666                 else:
--> 667                     self.new_state(RPDeviceConnected)
    668 
    669         except SegmentationNotSupported:

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in new_state(self, newstate)
    160         self._log.info('Changing device state to {}'.format(newstate))
    161         self.__class__ = newstate
--> 162         self._init_state()
    163 
    164 

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in _init_state(self)
    389 
    390     def _init_state(self):
--> 391         self._buildPointList()
    392 
    393 

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Device.py in _buildPointList(self)
    450         self._log.info('Device {}:[{}] found... building points list'.format(self.properties.device_id,self.properties.name))
    451         try:
--> 452             self.properties.objects_list, self.points = self._discoverPoints(self.custom_object_list)
    453             if self.properties.pollDelay > 0:
    454                 self.poll()

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py in _discoverPoints(self, custom_object_list)
    464 
    465                     presentValue=float(
--> 466                         self.read_single('{} {} presentValue '.format(point_type, point_address))),
    467                     units_state=self.read_single('{} {} units '.format(point_type, point_address)),
    468                     device=self))

ValueError: could not convert string to float: `

Many thanks for your attention on my issue.

ChristianTremblay, 2017-04-19:
We are close. There is at least one point not recognized.
Try to reduce your point list to minimum and add points until it fails. You will be able to know which one fails.

I suspect this one :  ('3:66', 81066).... but try one at a time.

ChristianTremblay, 2017-04-25:
Have you been able to make it work ?
Should I let this issue open or close it ?

MichelleNing, 2017-04-25:
Because I can not remove or delete BACnet objects from device, I cannot
continue to work on this issue.

You can close it. But I would like to say version 0.99.72 worked well on
the same device.

Could you please send me the version 0.99.72? From pip install
BAC0==0.99.72 or download from website, it doesn't work. I am appreciate if
you could send me the worked 0.99.72 version.

Thank you so much!

ChristianTremblay, 2017-04-25:
This is weird....
I will have to compare actual with 72...
This also means that you work with an older version of bacpypes right ?

I found the old archive which I think I can send you by email. I will need to work on my repo to tag it I can't do it right now...

Just send me an email and I will send you the files.

MichelleNing, 2017-04-25:
I installed BAC0 0.99.82 with new version of bacpypes first.Then I
uninstall BAC0 0.99.82 and install BAC 0.99.72.
After that, I even can't use BAC0.version.

My friends installed BAC 0.99.72 last summer and everything worked very
well. After he installed 0.99.82, he has the same issue. When he tried to
back to 0.99.72, it didn't work.

My email: tongn84@gmail.com.

Thank you so much!

ChristianTremblay, 2017-04-25:
Maybe there is a new feature in bacpypes 0.15 causing the issue with your device.
If reverting to 72, try with bacpypes 0.14

closed 2017-05-17 by ChristianTremblay