ChristianTremblay/BAC0 #40: Chart

ChristianTremblay, 2017-05-07:
@MichelleNing

I have a question for using chart.

lst1 = ['Room Temperature']
controller.chart(lst1, title = 'Room Temp')

C:\ProgramData\Anaconda3\lib\site-packages\bokeh\models\sources.py:89: BokehUserWarning: ColumnDataSource's columns must be of the same length
  lambda: warnings.warn("ColumnDataSource's columns must be of the same length", BokehUserWarning))

This warning message only can be stopped by force exit from bokeh server.

Do you have any idea about this?

ChristianTremblay, 2017-05-07:
Can you provide details about "Room Temperature" object. A print of it's history for example

MichelleNing, 2017-05-11:
It is an interesting result for history. I attached here:

>>> controller['Room Temperature'].history
2017-05-11 11:37:13.097500    25.888887
2017-05-11 11:38:01.398500    25.888887
2017-05-11 11:38:50.229500    25.944447
dtype: float64
>>> controller['Room Temperature'].history
2017-05-11 11:37:13.097500    25.888887
2017-05-11 11:38:01.398500    25.888887
2017-05-11 11:38:50.229500    25.944447
2017-05-11 11:38:57.478500    25.944447
dtype: float64
>>> controller['Room Temperature'].history
2017-05-11 11:37:13.097500    25.888887
2017-05-11 11:38:01.398500    25.888887
2017-05-11 11:38:50.229500    25.944447
2017-05-11 11:38:57.478500    25.944447
2017-05-11 11:39:03.629500    25.944447
dtype: float64
>>> controller['Room Temperature'].history
2017-05-11 11:37:13.097500    25.888887
2017-05-11 11:38:01.398500    25.888887
2017-05-11 11:38:50.229500    25.944447
2017-05-11 11:38:57.478500    25.944447
2017-05-11 11:39:03.629500    25.944447
2017-05-11 11:39:09.823500    26.000002
dtype: float64
>>> controller['Room Temperature'].history
2017-05-11 11:37:13.097500    25.888887
2017-05-11 11:38:01.398500    25.888887
2017-05-11 11:38:50.229500    25.944447
2017-05-11 11:38:57.478500    25.944447
2017-05-11 11:39:03.629500    25.944447
2017-05-11 11:39:09.823500    26.000002
2017-05-11 11:39:19.079500    26.055557
dtype: float64
>>> controller['Room Temperature'].history
2017-05-11 11:37:13.097500    25.888887
2017-05-11 11:38:01.398500    25.888887
2017-05-11 11:38:50.229500    25.944447
2017-05-11 11:38:57.478500    25.944447
2017-05-11 11:39:03.629500    25.944447
2017-05-11 11:39:09.823500    26.000002
2017-05-11 11:39:19.079500    26.055557
2017-05-11 11:39:29.304500    26.111111
dtype: float64

ChristianTremblay, 2017-05-12:
each time you call the point... it will read it ....AND  store the value... 
so each time you execute `controller['Room Temperature']` it will add a record

I'm unable to reproduce your bug

closed 2018-09-29 by ChristianTremblay