ChristianTremblay/BAC0 #41: Getting stuck on BAC0.connect()

miltoncolby, 2017-05-11:
Hello,

I'm struggling to get the most basic part of BAC0 working: program execution is getting stuck on the BAC0.connect() call. It outputs the IP it is using then hangs. 

> >> import BAC0
> >> bacnet = BAC0.connect()
Using ip : 192.168.2.55

..and then it waits indefinitely.

Has anyone else encountered this issue? 
I'd appreciate any help I can get, I imagine I'm missing something simple. Thanks.

ChristianTremblay, 2017-05-12:
If things work correctly, you should see :
```
In [1]: import BAC0
xlwings not installed. If using Windows or OSX, install to get more features.

In [2]: bacnet = BAC0.connect()
Using ip : 192.168.210.16
Click here to open Live Trending Web Page
http://localhost:5006/?bokeh-session-id=zYVBzGjBGVYx3Q9jZPR9nwIA6P0WeMU2rShrytRU
VZVV
```

Now maybe it's caused by bokeh trying to start the server.
Can you check what version of Bokeh is installed ?
```
import bokeh
bokeh.__version__
```
I'm actually using 0.12.2 and it works.... I'll test with newest...

You can also check if you are able to start the bokeh server
```
# In a Windows Shell
bokeh serve
```
It should tell you something
Here mine was already started
```
2017-05-11 22:11:15,098 Starting Bokeh server version 0.12.2
2017-05-11 22:11:15,100 Cannot start Bokeh server, port 5006 is already in use
```

miltoncolby, 2017-05-15:
@ChristianTremblay 
Thank you for the quick reply, here's what I found:

My bokeh version is 0.12.5

Calling "bokeh serve" outputs:
2017-05-15 08:17:19,081 Starting Bokeh server version 0.12.5
2017-05-15 08:17:19,081 Starting Bokeh server version 0.12.5

ChristianTremblay, 2017-05-15:
Correct. Please note I've seen a bug with charts with Bokeh 0.12.5
It will fail because I use VBOX instead of columns...
I patch will follow soon.

You could test the following : 
In a command window, start `bokeh serve`
The server will run on your machine.

Using another process (python shell or Jupyter Notebook) try to start BAC0. It is supposed to see that bokeh server is running and start... Maybe we'll find something else.

Be sure to turn on logging to see all messages : 
```
import logging
logging.getLogger().setLevel(logging.INFO)      # or DEBUG
```

miltoncolby, 2017-05-15:
Okay I did as you suggested and ran "bokeh serve" and while it was running did the following:

> >import logging
> >logging.getLogger().setLevel(logging.DEBUG)
> >import BAC0
> xlwings not installed. If using Windows or OSX, install to get more features.
> >bacnet = BAC0.connect()
> Using ip : 192.168.2.46
> Failed running bokeh serve
> Bokeh server already running

and once again the execution hangs here.

If my understanding of BAC0 is correct, the bokeh server is used for graphing trends and such. If this is the case, I don't believe I require that functionality; should I simply try working around it or try getting an older version of bokeh?

ChristianTremblay, 2017-05-15:
Just to be sure, what do you mean with `hangs here`. Do you mean Python tries to execute the line and it just stay there, computing and you loose access to the shell or Notebook ?
Are you unable to write anything ?
Are you using a Python Shell or a Jupyter Notebook ?

Because once you have defined the bacnet network using `bacnet = BAC0.connect()` you won't see anything if you don't write anything. There are not graphical interface.

You need to send commands like, for example : `bacnet.devices` will show you the devices found on your network.

miltoncolby, 2017-05-15:
I mean to say that it stays there, computing. I lose access to the program, I cannot write at all. The only thing that seems to have an effect is a keyboard interrupt (CTRL+C).

I've been using a Python Shell. I've tried both the python console in PyCharm environment and launching `python` inside the windows command line.

If I write a short script like:

> import BAC0
> bacnet = BAC0.connect()
> print(bacnet.devices)

the third line, `print(bacnet.devices)`, is never reached.

The only lead I've found is that when keyboard interrupting when running in the windows command line, it spits out an error trace before ending:

> C:\Users\miltonc\python>python
> Python 3.5.3 (v3.5.3:1880cb95a742, Jan 16 2017, 16:02:32) [MSC v.1900 64 bit (AMD64)] on win32
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import BAC0
> C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\BAC0\core\devices\Device.py:22: FutureWarning: The pandas.lib module is depr
> ecated and will be removed in a future version. These are private functions and can be accessed from pandas._libs.lib instead
>   from pandas.lib import Timestamp
> xlwings not installed. If using Windows or OSX, install to get more features.
> >>> bacnet = BAC0.connect()
> Using ip : 192.168.2.46
> Traceback (most recent call last):
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\packages\urllib3\connectionpool.py", line 386, in _make_req
> uest
>     httplib_response = conn.getresponse(buffering=True)
> TypeError: getresponse() got an unexpected keyword argument 'buffering'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\BAC0\scripts\ReadWriteScript.py", line 77, in __init__
>     self.start_bokeh()
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\BAC0\scripts\ReadWriteScript.py", line 89, in start_bokeh
>     requests.get('http://localhost:5006')
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\api.py", line 70, in get
>     return request('get', url, params=params, **kwargs)
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\api.py", line 56, in request
>     return session.request(method=method, url=url, **kwargs)
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\sessions.py", line 475, in request
>     resp = self.send(prep, **send_kwargs)
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\sessions.py", line 596, in send
>     r = adapter.send(request, **kwargs)
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\adapters.py", line 423, in send
>     timeout=timeout
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\packages\urllib3\connectionpool.py", line 595, in urlopen
>     chunked=chunked)
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\site-packages\requests\packages\urllib3\connectionpool.py", line 389, in _make_req
> uest
>     httplib_response = conn.getresponse()
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\http\client.py", line 1198, in getresponse
>     response.begin()
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\http\client.py", line 297, in begin
>     version, status, reason = self._read_status()
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\http\client.py", line 258, in _read_status
>     line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
>   File "C:\Users\miltonc\AppData\Local\Programs\Python\Python35\lib\socket.py", line 576, in readinto
>     return self._sock.recv_into(b)
> KeyboardInterrupt

I tried interpreting this error in hopes of finding a solution to the problem but didn't get far.
Sorry for the awful formatting.

Hope this helps! Thanks again for all the help thus far.

ChristianTremblay, 2017-05-15:
Thanks for the infos.

Are you able to test against the develop branch ?

I just made a modification to allow "not to use" bokeh
```
import BAC0
bacnet = BAC0.connect(bokeh_server=False)
```
clone the branch and use `python setup.py develop`

Keep me informed if it works

ChristianTremblay, 2017-05-15:
Also, can you provide the version of Tornado installed on your machine ? I've read something about bokeh and Tornado having some troubles actually...

miltoncolby, 2017-05-16:
I cloned the develop branch and ran the setup as instructed. **It seems to be working now!** It doesn't get stuck and it is detecting the simulated device I have running:

> -import BAC0
> xlwings not installed. If using Windows or OSX, install to get more features.
> C:\Users\miltonc\python\BAC0_dev\BAC0\BAC0\core\devices\Points.py:20: FutureWarning: The pandas.lib
> module is deprecated and will be removed in a future version. These are private functions and can be
>  accessed from pandas._libs.lib instead
>   from pandas.lib import Timestamp
> -bacnet = BAC0.connect(bokeh_server=False)
> Using ip : 192.168.2.46
> WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
> -bacnet.devices
>                                                                Manufacturer  \
> Name
> RoomController.Simulator  F. Chaxel, Thanks to Morten Kvistgaard, MIT li...
> 
>                                                          Address   Device ID
> Name
> RoomController.Simulator  0xc0a8022edccc          66

My Tornado version: `'4.5.1'`

Thank you for resolving my issue. You've been quick to reply and have provided excellent help. If someone in my position needed bokeh then perhaps this wouldn't be ideal but it is the perfect solution for me.

If you have any other questions or if I can help test something else just let me know. 
Thanks again!

ChristianTremblay, 2017-05-16:
Here is the link for the issue with Bokeh and tornado 4.5
https://github.com/bokeh/bokeh/issues/6152

ChristianTremblay, 2017-05-17:
I keep this open until bokeh is repaired.
Will change dependency to higher bokeh version when fixed

ChristianTremblay, 2017-06-14:
@miltoncolby Can you try the newset bokeh version.... they are saying the bug as been repaired
https://github.com/bokeh/bokeh/issues/6152

miltoncolby, 2017-06-14:
@ChristianTremblay Just ran a basic test, see below.

> import bokeh
> bokeh.__version__
> '0.12.6'
> import tornado
> tornado.version
> '4.5.1'
> import BAC0
> xlwings not installed. If using Windows or OSX, install to get more features.
> C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.2\helpers\pydev\_pydev_bundle\pydev_import_hook.py:21: FutureWarning: The pandas.lib module is deprecated and will be removed in a future version. These are private functions and can be accessed from pandas._libs.lib instead
>   module = self._system_import(name, *args, **kwargs)
> BAC0.version
> '0.99.102'
> bacnet = BAC0.connect()
> Using ip : 192.168.2.92
> Click here to open Live Trending Web Page
> http://localhost:5006/?bokeh-session-id=0xdZBeDlXQRrMfvJhcUz9nMr5bqFEC2ctcnsF4YOtPZm
> bacnet.disconnect()
> Stopping BACnet stack
> BACnet stopped

Seems to be fixed. I'm happy to help if you feel it requires further testing.

ChristianTremblay, 2017-06-14:
Great ! Case closed

closed 2017-06-14 by ChristianTremblay