ChristianTremblay/BAC0 #43: bacnet.devices error

henkwitte, 2017-06-13:
Hello, 

I am trying to get BAC0 to work on beaglebone (debian 8, jessie). I ran into the bokeh issue described in another post and installed the development branch. I can then run the bacnet=BAC0.connect(bokeh_server=False) command.

After this I run the bacnet.devices command and get the following error:

>>> bacnet.devices
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/groenhol/BAC0/BAC0/scripts/BasicScript.py", line 193, in devices
    return pd.DataFrame(lst, columns=['Name', 'Manufacturer', 'Address',' Device ID']).set_index('Name').sort_values('Address')
  File "/home/groenhol/miniconda3/lib/python3.4/site-packages/pandas/core/generic.py", line 2150, in __getattr__
    (type(self).__name__, name))
AttributeError: 'DataFrame' object has no attribute 'sort_values'

Sorry if the fault is something obvious - I am new to python and its modules and installing the complete environment on the beaglebone has been an experience...

Any idea what I can do about this?

Thanks!

ChristianTremblay, 2017-06-14:
It looks like a Pandas issue. DataFrame are Pandas objects. 
sort_values have been added in 0.17 and miniconda seems to give access to 0.16
I'll make some test on a Raspberry Pi... it's the closest thing to BeagleBone I have
On your side, you could check if you can update pandas using `pip install pandas --upgrade`

ChristianTremblay, 2017-06-14:
I've been able to upgrade pandas to 0.20 on the Raspberry Pi 3. It took about 1 hour to compile, no joke... But it works...
Now I could also modify the code to handle version without sort_values... I think it would be better for older or smaller devices.

ChristianTremblay, 2017-06-14:
Please try to pull the develop branch and tell me if it works for you. I modified the code to use sort_values "if possible"
I've also made a modification related to pandas Timestamp... you will be able to confirm it still work with an old pandas version (0.16)

henkwitte, 2017-06-14:
Hi Cristian:

1. Checked pandas version: 0.16.2
2. Update pandas - this fails for me as gcc gives an internal compile error. This is why I had to use conda/miniconda to install (and then as you say I am limited to 0.16)
3. I run git pull (I see develop branch being updated) and then I run python setup.py develop (which is my active branch)

now however, the bacnet=BAC0.connect(bokeh_server=False)  does not work anymore. 

I guess I am doing something wrong here. This evening I will try to go back to my saved boot image of last night and see if that works. 

Thanks for your help so far, I will update later if it works for me.

Henk

henkwitte, 2017-06-14:
Hi, Ok - I noted the bokeh issue has been solved. Now trying to update bokeh without breaking the other stuff (e.g. trying to install pandas which gives gcc error). Once bokeh sorted will retry development branch.

henk

henkwitte, 2017-06-14:
So, I am a bit stuck. You removed the workaround for bokeh but I cannot find the 12.6 version for the armv7l and therfore cannot check the develop branch to deal with the pandas issue.
Any suggestions?

ChristianTremblay, 2017-06-14:
The bokeh_server argument is still in the dev branch
https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/scripts/ReadWriteScript.py#L66
If it doesn't work.. there is something wrong

ChristianTremblay, 2017-06-14:
Regarding bokeh, are you able to `pip install bokeh` ?
On my side I have been able to install it on a Raspberry Pi running "pixel" the latest update.

henkwitte, 2017-06-15:
Installing bokeh with pip also tries to install pandas, which gives an gcc error.

However, I found that it depends if I run python in the BAC0 directory or not, if I run it outside and import BAC0 I get an error on the bacnet = BAC0.connect(bokeh_server=False) command, if I run it inside the BAC0 directory (develop branch) it does not give the fault,

Moreover, bacnet.devices now works!

Great. I think there is something about the git branches I do not understand or a fault in my setup.

Thanks ! Now the next step: reading the bacnet points.....

ChristianTremblay, 2017-06-16:
You are probably facing a path issue with the install....
I would uninstall BAC0 (`pip uninstal BAC0`, repeat until no version available....
Then you go in the BAC0 folder and type `python setup.py install`

You could also have used `pip setup.py develop` to replace the path of BAC0 for this folder.... your choice.

henkwitte, 2017-06-20:
Hi Christian,

uninstalled BAC0 as you suggested and reinstalled, both master and develop. I was also able to upgrade to bokeh 0.12.6 - but have not yet tried recompiling pandas.

Any case, BAC0.version is now 0.99.103

If I run the BAC0.device on the PCO carel controller I get:

>>> test=BAC0.device('192.168.178.101',77000,bacnet)
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...

Will try to test also against the Jace (which we know is working) but tomorrow I am not in the office.

Henk

henkwitte, 2017-06-22:
I decided to try the beaglebone with ubuntu 16 in stead of debian 8 (jessie). Installed the ubuntu image, installed miniconda. Then installed bokeh with pip and installed BAC0.

With BAC0 now pandas is updated and installes correctly, I can read the bacnet datapoints from a Jace controller.

Summary: use ubuntu on the beaglebone black rev C and not debian 8 (jessie)

Henk

closed 2017-06-22 by henkwitte