ChristianTremblay/BAC0 #44: MyController.points issue

henkwitte, 2017-06-19:
Hi Christian,

After your suggestion to wait for some time before issuing the ~.getpoints command, I tried waiting for 15 minutes after the BAC0.device() command, but with the same result:

PS, this is on the master branch

>>> bacnet.devices
            Manufacturer         Address   Device ID
Name
Norwich_100      Tridium  192.168.25.101         100
>>> Norwich=BAC0.device('192.168.25.101',100,bacnet)
>>> Norwich.points
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\henkwitte.GROENHOLLAND\AppData\Local\Continuum\Anaconda3\lib\site-packages\BAC0\core\devices\Points.py", line 410, in __repr__
    return '%s : %.2f %s' % (self.properties.name, self.history.dropna().iloc[-1], self.properties.units_state)
  File "C:\Users\henkwitte.GROENHOLLAND\AppData\Local\Continuum\Anaconda3\lib\site-packages\pandas\core\indexing.py", line 1328, in __getitem__
    return self._getitem_axis(key, axis=0)
  File "C:\Users\henkwitte.GROENHOLLAND\AppData\Local\Continuum\Anaconda3\lib\site-packages\pandas\core\indexing.py", line 1749, in _getitem_axis
    self._is_valid_integer(key, axis)
  File "C:\Users\henkwitte.GROENHOLLAND\AppData\Local\Continuum\Anaconda3\lib\site-packages\pandas\core\indexing.py", line 1638, in _is_valid_integer
    raise IndexError("single positional indexer is out-of-bounds")
IndexError: single positional indexer is out-of-bounds

ChristianTremblay, 2017-06-19:
As this is a Jace.... point list comes from a export table.
Can you share the table ?
Or limit the table to 1 entry so we can narrow the search scope ?
We could also use this validation to put our finger on a failing point (if one in particular)
```
for point in Norwich.points:
   print(point)
```
if the pandas part fails, you can verify that at least points name are ok
```
for point in Norwich.points:
    print(point.properties.name)
```

henkwitte, 2017-06-19:
Hi Christian,

That works - in the original table there is apparently one point that breaks the loop. Putting one new point in the table, updating the BAC0.connect and then running the 

for point in Norwich.points:
   print(point)

shows:

CalcMonth.testBCvalue : 15.50 noUnits

Thanks, will now start more tests with the beaglebone as well.

Henk

ChristianTremblay, 2017-06-19:
Great ! One more step

closed 2017-06-19 by ChristianTremblay

henkwitte, 2017-06-30:
Issue with Carel PCO controller also solved:

On the beaglebone with ubuntu and running the BAC0 development branch, following Christian's procudure by defining an object list to reduce the number of points retrieved:

import BAC0
BAC0.version (= 0.99.105) _(just to check)_
bacnet=BAC0.connect(bokeh_server=False)
my_ob_list=[('file',1),('analogValue',100062)]
fx2=BAC0.device('192.168.25.24',77000,bacnet,object_list=my_ob_list)
fx2.points

Shows the actual value of the point:
[SET_ACS_TANK : 45.00 noUnits]

Problem was that the Carel device has a great number of points (> 6000!) which seems to cause issues with retrieving the data.

Henk

ChristianTremblay, 2017-07-05:
Carel PCOs are somewhat, exagerating.... :-)
Glad to see it fixed the trouble also !