ChristianTremblay/BAC0 #45: Unable to Install BAC0 on windows

spadilam, 2017-06-21:
Team,

I am trying to use python 3.6 version and when trying to install BACO. I am facing problem with the installation.

Could you please help us out in this process.

Regards,
Srinivas Padilam.

henkwitte, 2017-06-22:
Install anaconda (https://www.continuum.io/downloads) open an anaconda prompt (see start menu, anaconda3, anaconda prompt) and run pip install BAC0

Worked for me!

Henk

ChristianTremblay, 2017-06-23:
If @henkwitte solution, please provide more details.
But I'm pretty sure Anaconda will work.

spadilam, 2017-06-24:
Hi Team,
Do I need to install anaconda as well. 
 I trying to use the libraries directly in python3.6 and installed bacpypes and bokeh packages ...
After that when trying to intall the BAC0 I am getting issue with VBox ...

Please suggest me the solution. I want to use with solution without anaconda package.

ChristianTremblay, 2017-06-24:
It's due to the new bokeh version. I'll put a new version on pypi soon. 

In the meantime you can use the develop branch and it will work.

ChristianTremblay, 2017-10-12:
@spadilam  Have you had the chance to make it work ?
If so, I'll close the issue.

closed 2018-01-19 by ChristianTremblay

gandhamyamini467, 2019-03-27:
Hello im not able to work it out?
Can you please suggest me how to install BAC0.I have installed anaconda too,but of no use.Please help me with this
[image]

ChristianTremblay, 2019-03-30:
run `python setup.py install` in BAC0-master.
Or simply use `pip install BAC0` anywhere if you don' need to modify code.