ChristianTremblay/BAC0 #46: minor issue, bokeh server already running

henkwitte, 2017-06-22:
Hi, noted this on debian and now also on ubuntu (beaglebone black):

If I run import BAC0 and then bacnet=BAC0.connect() I get an error "bokeh server not running". BAC0 fails

If I run first import bokeh, then import BAC0 and BAC0.connect() I get an error "Failed running bokeh server,Bokeh server already running" . BAC0 is running ok after this, so it does not seem to affect the software.

Henk

ChristianTremblay, 2017-06-22:
Just to add some details.
`bokeh serve` is started as a subprocess inside BAC0. 
I've never been able to start the server "inside" BAC0 via a standard function. As the server rely on Tornado, the server needs to be on the main thread... 
It was way over my skill to find a clean solution so I ended up with a subprocess.
That said, once the server is started in its own process, it will continue to run even if BAC0 stops. This is why we see the warning message about "Bokeh server already running".

What bothers me in your issue is the fact that importing bokeh before BAC0 in your script seems to make a difference.

I guess I will have to do tests. Here I can make tests on a Raspberry Pi running Raspbian... We'll see if I can reproduce.

@henkwitte are you sure there is no path issue that could explain your problem ? I mean, import bokeh is part of the startup steps of BAC0 when connecting...

henkwitte, 2017-06-23:
Hi, I noticed this on a completely fresh install on ubuntu TL16.04.

Maybe I need to give the bokeh subprocess more time to start - I will test that.

Any way in which I can check the path?  

PS: I intend to run the bacnet connections "for ever" as a controller for a system, I guess it would be better to not start bokeh and other trending modules? Any pointers on what to do to make that robust (of course, I will go through the scripts as well...).

Henk

ChristianTremblay, 2017-06-23:
To run forever, you should use 
`bokeh_server = False` option
and when you define a device, add the argument `poll=0` This will deactivate polling from every points in the device.

You could also use a new feature (in the develop branch) that allow to specify the point list when defining a device. This would allow you to keep the polling feature active while restricting the point list.

Here is an extract from the doc : 

```
**Object List**

By default, BAC0 will read the object list from the controller and define every
points found inside the device as points. This behaviour may not be optimal in
all use cases. BAC0 allows you to provide a custom object list when creating the
device.

To do so, use this syntax::    

    # Define your own list
    my_obj_list = [('file', 1),
                 ('analogInput', 2),
                 ('analogInput', 3),
                 ('analogInput', 5),
                 ('analogInput', 4),
                 ('analogInput', 0),
                 ('analogInput', 1)]
    
    # Provide it as an argument               
    fx = BAC0.device('2:5',5,bacnet, object_list = my_obj_list)
```

ChristianTremblay, 2017-06-26:
@henkwitte I found out yesterday about "berryconda" which allow Python 3.6 to run on the Raspberry Pi. I guess you could check that out for the Beaglebone.

I've installed it on my Raspberry Pi and it works great.

https://github.com/jjhelmus/berryconda

henkwitte, 2017-06-26:
Hi - thanks for that! I will check it out on debian jessie a soon as i find time, for now I will keep with ubuntu as that works OK. I will update this thread when I test berryconda with debian.
Henk

henkwitte, 2017-06-29:
Hi,

I have tried running the install for debian on berryconda - however it does not seem to have conda installed so I install with pip. However, I again get errors on installing BAC0 related to the compiling of pandas in python 3 (for python 2.7 it runs ok, although taking a lot of time).  Installing and compiling with pip on python 3 takes several hours....and then fails. I suppose I could start over with a completely fresh install of Debian and then test as I think my environment is not clean anymore. 

Unfortunately I am stressed for time and with the ubuntu operating system I do not have these issues.

For now, I will need to leave testing debian to a later date.

Henk

ChristianTremblay, 2017-07-05:
Will close then...

closed 2017-07-05 by ChristianTremblay