ChristianTremblay/BAC0 #47: Connecting to device behind BACnet router

SmilingWorlock, 2017-07-19:
Hello,

I am trying to connect to a device behind a BACnet router. I know the following: 
The router has the IP - Adress 192.168.0.4
The Device ID of the router is 1050658
The Device ID I want to connect to is 2131969

I can connect to the router and get it's empty object list. I can however not find a way to connect to the actual device I want to connect to the actual device. 
The router is BACnet/IP to LON. 
Some addded problems: I connect from a different network, where broadcasts are not possible. YABE on a maintanance PC in the 192.168.0.1/24 network shows "1 1 via  192.168.0.4", so the device is there. I cannot use YABE myself.

Is there anyway to establish this connection?

ChristianTremblay, 2017-08-08:
Humm....This is something I never tried.

When you declare your IP, do you specify the subnet (/24 for example) ?

Can your router be configured to add a foreign device pointing to your computer. Broadcast messages could then be sent to you by the router.

Refer to your router documentation for that part, it should be explained.

ChristianTremblay, 2018-03-06:
The part regarding different subnet will be supported using the foreigndevice feature actually implemented in the develop branch since v.0.9.909.

As this thread has not been fed for a long time, I will close it.

closed 2018-03-06 by ChristianTremblay