ChristianTremblay/BAC0 #49: Error when running in module?

henkwitte, 2017-08-04:
Hi,

If I open a python shell and enter these commands all runs well:

```
>>> import BAC0
>>> bacnet=BAC0.connect(bokeh_server=False)
Starting BAC0 version 0.99.105
Using ip : 192.168.25.38
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
>>> bacnet.devices
             Manufacturer         Address   Device ID
Name
Norwich_100       Tridium  192.168.25.101         100
pCOWeb77000  Carel S.p.A.   192.168.25.22       77000
>>> cntry_ecy=BAC0.device('192.168.25.101',100,bacnet)
>>> for point in cntry_ecy.points:
...  print(point.value)
...
15.5
>>>
```

However, if I put the commands in a test.py file and run this with "python test.py":

```
print("Starting bacnet processing")
import BAC0
bacnet=BAC0.connect(bokeh_server=False)
bacnet.devices
cntry_ecy=BAC0.device('192.168.25.101',100,bacnet)
for point in cntry_ecy.points:
 print(point.value)
```

an error is raised:

```
Starting BAC0 version 0.99.105
Using ip : 192.168.25.38
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
Exception in thread rpm_poll:
Traceback (most recent call last):
  File "/home/groenhol/miniconda3/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/home/groenhol/miniconda3/lib/python3.4/site-packages/BAC0-0.99.105-py3.4.egg/BAC0/tasks/TaskManager.py", line 44, in run
    self.process()
  File "/home/groenhol/miniconda3/lib/python3.4/site-packages/BAC0-0.99.105-py3.4.egg/BAC0/tasks/TaskManager.py", line 50, in process
    self.task()
  File "/home/groenhol/miniconda3/lib/python3.4/site-packages/BAC0-0.99.105-py3.4.egg/BAC0/tasks/Poll.py", line 71, in task
    self._device.read_multiple(list(self._device.points_name), points_per_request=25)
  File "/home/groenhol/miniconda3/lib/python3.4/site-packages/BAC0-0.99.105-py3.4.egg/BAC0/core/devices/mixins/read_mixin.py", line 134, in read_multiple
    points_values = zip(big_request[1][i:i + len(val)], val)
TypeError: object of type 'NoneType' has no len()
```

ChristianTremblay, 2017-08-08:
This is weird. I'll make tests on my side.

ChristianTremblay, 2017-08-08:
I suspect something wrong with one point.
I'm able to run modules from the cmd line on my computer,

Please try using : 
```
for point in cntry_ecy.points:
 print(point)
```

It should give you something before crashing and help you find the "not correctly handled" point

See what I get
```
(C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3) C:\0programmes\g
ithub>python bac0_modules.py
Starting bacnet processing
Using ip : 192.168.111.31
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will
 not work
EA-VP : 55.00 pascals
DPR-O : 100.00 percent
CLG-MAXFLOW : 488.00 litersPerSecond
CLGOCC-MINFLOW : 111.00 litersPerSecond
HTGOCC-MINFLOW : 111.00 litersPerSecond
SA-AREA : 0.03 squareMeters
SA-KFACTOR : 2.00 noUnits
OCCMODE-BYPASSTIME : 3600.00 seconds
EA-F : 217.10 litersPerSecond
Exhaust Damper Control Output ERROR : 70.90 noUnits
Exhaust Damper Control Output ABSERROR : 70.90 noUnits
Exhaust Damper Control Output ABSEFFORT : 0.00 noUnits
Exhaust Damper Control Output EWMA : 100.00 noUnits
EFFEXHAUSTAIRFLOW-SP : 288.00 noUnits
Room Area : 334.00 squareFeet
Room Height : 9.00 feet
ACHmin_UnOccupied : 4.00 noUnits
ACHmin_Occupied : 8.00 noUnits
Airflow for ACH_min : 189.16 litersPerSecond
UnOccupied : 100.00 noUnits
Occupied : 100.00 noUnits
FLOWDIFF-SP : 50.00 litersPerSecond
SUPPLYFLOW-DEMAND : 100.00 noUnits
EFFHOOD-FLOW : 200.00 noUnits
SUPPLYVAVMAX-AIRFLOW : 438.00 noUnits
HOODVAVMIN-AIRFLOW : 111.00 noUnits
OCC-SCHEDULE : Occupied
EFF-OCC : Occupied
SF-S : On
Exhaust Damper Control Output SL-OUTSTATE : Control
HOOD_OCC-SENSOR : Not Set
TUNING-RESET : False
AUTOCAL-C : False
```

henkwitte, 2017-08-08:
Hi,

For testing only one point defined. If I run "print(point)", it ran fine:

`print("Starting bacnet processing")
import BAC0
bacnet=BAC0.connect(bokeh_server=False)
bacnet.devices
cntry_ecy=BAC0.device('192.168.25.101',100,bacnet)
for point in cntry_ecy.points:
    print(point.value)
`

Which shows:

`
Starting bacnet processing
Starting BAC0 version 0.99.105
Using ip : 192.168.25.38
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
CalcMonth.testBCvalue : 15.50 noUnits
Done bacnet processing`

If I then run print(point.value) I again receive an error. 

If I enter the commands directly in the python console it runs fine, with the same point and gives the value 15.5 :

```
>>> import BAC0
>>> bacnet=BAC0.connect(bokeh_server=False)
Starting BAC0 version 0.99.105
Using ip : 192.168.25.38
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
>>> cntry_ecy=BAC0.device('192.168.25.101',100,bacnet)
>>> for point in cntry_ecy.points:
...     print(point.value)
...
15.5
>>>
```

But running this script as _python test.py_ again fails.

After the failure I cannot run the original command anymore either, when I run the script with print(point) I also get an error: BAC0.core.io.IOExceptions.ApplicationNotStarted: BACnet stack not running - use startApp() 

So additionally, is there a way to gracefully recover BAC0 from a fault (i.e. without reboot)?

Henk

ChristianTremblay, 2017-08-09:
From my side, using point.value works : 

```
(C:\Users\ctremblay.SERVISYS\AppData\Local\Continuum\Anaconda3) C:\0programmes\g
ithub>python bac0_modules.py
Starting bacnet processing
Using ip : 192.168.111.31
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will
 not work
55.0
100.0
488.0
111.0
111.0
0.03240000084042549
2.0
3600.0
217.10012817382812
70.89884948730469
70.90089416503906
3.1674988154500886e-36
99.99897766113281
288.0
334.0
9.0
4.0
8.0
189.1565704345703
100.0
100.0
50.0
100.0
200.0
438.0
111.0
1
1
2
3
5
inactive
inactive
```

To recover from a fault, you will probably need to catch the exception and run `bacnet.disconnect()` to free the socket.

I mostly work in Jupyter Notebook and when not using try block, I simply restart the kernel.

I guess killing python processes should work also, but it is violent...

henkwitte, 2017-08-10:
I have been trying different things and the results are not very clear, but:

1) print(point.value) on the EC Bos controller sometimes works, I have not been able to reproduce exactly when it works and when not, seems to be random. 

2) print(point.value) on the PCO carel controller always works. It gives a warning:

WARNING:BAC0.core.devices.Device:Device too slow, use single points polling if needed
WARNING:BAC0.core.devices.Device:Points will be read once...

The fault occurs in mixin.py at line 134:

_points_values = zip(big_request[1][i:i + len(val)], val)_

Here it seems val is an object of NoneType which has no len.

Any way in which I could further debug this?

ChristianTremblay, 2017-08-15:
To get rid of the warning, when you define the device, you can use the argument `segmentation_supported=False` (http://bac0.readthedocs.io/en/latest/connect.html#segmentation)

I must say that Carel PCO are not always easy to work with. Too much variables... a lot just not defined....

To troubleshoot, you can use some tricks  : 
```
# This will read the object list... so you can see if there's something weird
bacnet.read('192.168.25.101 device 100 objectList')
# You can then make some tests like
bacnet.read('192.168.25.101 analogInput 1001 presentValue')

# Using bacnet.read, you are working directly on the stack... no device implied.
```

Also, for PCO, you should really define an object_list you want to use to define the device.
There are a lot of unused variables that will just pollute everything.

Refer to : http://bac0.readthedocs.io/en/latest/connect.html#object-list

henkwitte, 2017-08-17:
Hi,

I ran a test with a simple script, only trying to list the devices on the network, and i have tried a test with reading points using an object list (perhaps this is why in the previous example the pco did work and the ecy controller not).

I have also switched to a production ECY-303 controller (which will be used alongside the carel PCO).

If I use object lists it runs OK, so that seems a good workaround. However, the bacnet.device gives an empty dataframe when run from the bash command line (python bacnet_data.py)

PS, I note that the device.connect "hangs" when the points are not published on the controller, is it possible to "try" this or have the command timeout? Also, as bacnet.devices does not work in a module, is there another way to test if controllers are online?

The code is:

```
import BAC0

flag_debug = True

def bacnet_connect():

    bacnet = BAC0.connect(bokeh_server=False)
    return bacnet

def exit(bacnet):
    bacnet.disconnect()
    if(flag_debug):
        print("stopping bacnet stack")

def list(bacnet):
    print(bacnet.devices)

def collect_data(bacnet):
    if(flag_debug):
        print("connecting to device")
    my_obj_list = [('file', 1),('analogValue', 1)]
    cntry_ecy=BAC0.device('192.168.178.105',178105,bacnet,object_list = my_obj_list,segmentation_supported=False)
    if(flag_debug):
        print("printing points")
    print(cntry_ecy.points)
    print(cntry_ecy['SpaceTemp'].value)

if __name__ == '__main__':
    if(flag_debug):
        print("starting bacnet stack")
    bacnet = bacnet_connect()
    list(bacnet)
    collect_data(bacnet)
    exit(bacnet)
```

Result when functions called in python shell:

```
>>> import BAC0
>>> bacnet = BAC0.connect(bokeh_server=False)
Starting BAC0 version 0.99.105
Using ip : 192.168.178.44
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
bacnet=bacnet_data.bacnet_connect()
Starting BAC0 version 0.99.105
Using ip : 192.168.178.44
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
>>> bacnet_data.list(bacnet)
                          Manufacturer          Address   Device ID
Name
ECY-303-8A9B0B  Distech Controls, Inc.  192.168.178.105      178105
>>> my_obj_list = [('file', 1),('analogValue', 1)]
>>> cntry_ecy=BAC0.device('192.168.178.105',178105,bacnet,object_list = my_obj_list,segmentation_supported=False)
WARNING:BAC0.core.devices.Device:Device too slow, use single points polling if needed
WARNING:BAC0.core.devices.Device:Points will be read once...
>>> cntry_ecy.points
[SpaceTemp : 25.45 degreesCelsius]
>>> cntry_ecy['SpaceTemp']
SpaceTemp : 25.48 degreesCelsius
```

And when running in the bash shell with python bacnet_data.py:

```
groenhol@arm:~/py_control$ python bacnet_data.py
starting bacnet stack
Starting BAC0 version 0.99.105
Using ip : 192.168.178.44
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will not work
Empty DataFrame
Columns: [Manufacturer, Address,  Device ID]
Index: []
connecting to device
WARNING:BAC0.core.devices.Device:Device too slow, use single points polling if needed
WARNING:BAC0.core.devices.Device:Points will be read once...
printing points
[SpaceTemp : 23.25 degreesCelsius]
23.245038986206055
Stopping BACnet stack
BACnet stopped
stopping bacnet stack

```

ChristianTremblay, 2017-10-02:
What if you would send output to a log file ?
It may be a Pandas issue ?

dayanand4u, 2018-02-28:
Hi Tremblay,
                   I am facing below issue while reading device.
>>> abc=BAC0.device('192.168.168.8',240002,bacnet)
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D0668>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D07B8>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D51D0>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D5320>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D5CF8>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D5E48>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36E5828>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36E5978>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36E5AC8>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36E5C18>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36E7BE0>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36E7E48>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D0DA0>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D0EF0>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D0860>
 ! <bacpypes.basetypes.ErrorType object at 0x0000002AA36D0E10>
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 148, in
 __init__
    self.new_state(DeviceDisconnected)
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 159, in
 new_state
    self._init_state()
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 635, in
 _init_state
    self.connect()
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 661, in
 connect
    self.new_state(RPMDeviceConnected)
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 159, in
 new_state
    self._init_state()
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 387, in
 _init_state
    self._buildPointList()
  File "C:\Python36\lib\site-packages\BAC0\core\devices\Device.py", line 448, in
 _buildPointList
    self.properties.objects_list, self.points = self._discoverPoints(self.custom
_object_list)
  File "C:\Python36\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", l
ine 293, in _discoverPoints
    self._log.warning('Cannot add {} / {]'.format(point_type, point_address))
ValueError: expected '}' before end of string

What would be the problem and solution.

Could you please provide  bacnet.write command sample script for update present Val of BinaryValu

ChristianTremblay, 2018-02-28:
@dayanand4u 
There is a typo in the code that has been corrected in the develop branch.
Can you try ?
(you may need to uninstall BAC0 first using pip uninstall BAC0 until there is nothing left....)
This branch contains a lot of improvements to deal with devices creation.

Regarding your other question : 
```
bacnet.write('3:5 binaryValue 10140 presentValue active - 8') 
```
would write BV:10140 presentValue at priority 8

Please start a new issue next time to help me maintain things clear

dayanand4u, 2018-03-01:
Thanks Tremblay i am able to resolve the faced issue by taking read_mixin.py file from developer branch

What  bacnet services it is supporting.. I am looking for Atomic Read File and Atomic Write File...Is it supporting.

ChristianTremblay, 2018-09-29:
Files are not supported in BAC0 actually

closed 2018-09-29 by ChristianTremblay