ChristianTremblay/BAC0 #50: issue writing point on distech ecy controller

henkwitte, 2017-09-21:
Hi,

If I try to write to a analogValue on a  eclypse controller the first attempt fails. If I interrupt and set the point to 'auto' I get an error message, but after this I can write to the point.

Also, to write I have to put the value between quotes.

```
>>> my_controller['variable_1'] = '4'
^CERROR:BAC0.core.devices.Device:Value must be numeric
>>> my_controller['variable_1'] = 'auto'
ERROR:BAC0.core.devices.Device:Value was not simulated or overridden, cannot release to auto
>>> my_controller['variable_1'] = '4' # now the value in the controller changes
```

fgervais, 2017-09-28:
I did a small test with an S1000 and it seems to work fine. Maybe you can check how your setup differs from mine.

I'm running this simple gfx code:
[image]

And I'm setting AV2 through python so it starts adding:
```python
import BAC0
import time

import sys

import os

bacnet = BAC0.connect("172.16.8.100")

try:
    testd = BAC0.device('172.16.11.54', 11054, bacnet)

    print(testd['Analog Value 1'])
    print(testd['Analog Value 2'])

    testd['Analog Value 2'] = 1

    time.sleep(3)

    testd.chart(['Analog Value 1'])

    while True:
        time.sleep(1)

except KeyboardInterrupt:
    bacnet.disconnect()
    try:
        sys.exit(0)
    except SystemExit:
        os._exit(0)
```

I'm running using:
python 3.5
BAC0==0.99.101
bacpypes==0.16.3
bokeh==0.12.9
pandas==0.20.3

henkwitte, 2017-09-28:
Hi Francois,

I am running 
python 3.4.3
BAC0 0.99.105
bacpypes  0.16.1

I do not use bokeh or pandas (I am starting with bokeh_server=False)

I am not sure if there have been changes (you are running somewhat newer versions). I will check that later (this is currently on a production platform).

henkwitte, 2017-09-30:
I have duplicated your gfx code on anm ECY S1000 and also upgraded bacpypes to 0.16.3. I still have the same issue, I cannot write to the point unless I first issue an 'auto' (which gives an error).

I will try to upgrade to python 3.5, but the miniconda available for my Beaglebone (arm) does not yet have it as far as I can tell.

I will be abroad most of next week so further testing will have to wait for a bit.

Thanks for your help so far!

ChristianTremblay, 2017-10-02:
The idea behind the setters functions of BAC0 via `device['variable'] = value` is : 

if the point is an AnalogValue, we will try to write to the presentValue. The value sent, should be a number (not a string) but only for "clarity purposes". In fact, because I'm using bacpypes, everything will be turned into a big string of text to be sent to the network. 

An analogValue is not an AnalogInput. This is why I treat them differently. Typically, analogInput are linked to physical points. This means that we should not in normal time, write to them. BAC0 will interpret the need to write to an analogInput as a request to "override" the value of the sensor. For that, BAC0 will first use the òut_of_service` flag to disconnect the sensor from the variable.... then BAC0 will write to the presentValue, resulting in a "override" of the sensor value.

The `auto` value will be used in that context to reset the `out_of_service` flag to false and so reconnect the sensor to the analogInput.

The same will be done for analogOutput, binaryInput/Output, multistateInput/Output for the same reasons.

Now all this doesn't explain why @henkwitte is getting errors.

I would explore direct command using :
```
bacnet.write("address analogValue 1 presentValue 100 16")
```
Using this, you in fact use bacpypes directly to write to the point. It is then possible to make all kind of verifications...

henkwitte, 2017-10-02:
Hi Al,

I will run these tests and I will also see  if I can log the actual bacnet traffic (with wireshark). However, I am abroad the rest of the week so it will take a bit of time before I can run the test.

@Cristian: you just want the console log or do I need to run some debug log?

Henk

henkwitte, 2017-10-16:
I have now been testing with the bacnet.write command and that seems to work fine. 
Strangely the original  testd['Analog Value 1'] = 1.0 also works when I run it after the bacnet.write command.

I am trying to find out under what circumstances the faults I noted earlier occur, I hope I can find that out running a few more tests

henkwitte, 2017-10-17:
I have been trying a few more tests, running commands from the python console and from a python script. So far the statement

testd['Analog Value 2'] = 1

still gives the same faults, I have not been able to find out why.

testd['Analog Value 2'] = '1'

sometimes works

The command

bacnet.write("192.168.178.98 analogValue 1 presentValue 1")

works and seems stable. 

bacnet.write("192.168.178.98 analogValue 1 presentValue " + str(set_value))

allows me to use a variable that i can manipulate

This is all on the beaglebone with Ubuntu 16

I have tried to trace with wireshark, but it seems the error already occurs in the script before there is any bacnet activity on the network as I do not see any traffic (I do see the inital whois etc. communication on the connect command).

ChristianTremblay, 2018-09-28:
It is a really old issue that may have been fixed by current version. @henkwitte can we close ?

henkwitte, 2018-10-04:
Dear Christian,

Closing the thread is fine!

Henk

closed 2018-10-04 by ChristianTremblay