ChristianTremblay/BAC0 #52: Trying to get started

estutzenberger, 2017-11-01:
Hi,

I'm trying to check out BAC0. It seemed to install successfully but, I'm hitting snag. I installed BAC0 on OS X and Linux and I'm running into the following issue on both systems:

```
Python 3.5.2 (default, Sep 14 2017, 22:51:06) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import BAC0
/home/xxx/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py:22: FutureWarning: The pandas.lib module is deprecated and will be removed in a future version. These are private functions and can be accessed from pandas._libs.lib instead
  from pandas.lib import Timestamp
xlwings not installed. If using Windows or OSX, install to get more features.
cannot import name 'gridplot'
>>> bacnet = BAC0.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'BAC0' has no attribute 'connect'
```

I'm wondering if the problem is related to `cannot import name gridplot`.

Any ideas?

ChristianTremblay, 2017-11-02:
Maybe, please try running 
```
bacnet = BAC0.connect(bokeh_server = False)
```

you may also want to try the develop branch (`python setup.py install`in the cloned folder)
Please note that the new Bokeh version 0.12.10 is giving me a hard time actually....

ChristianTremblay, 2018-02-17:
Using the develop branch will fix this issue....next release also indeed. The lite version will not need pandas, bokeh or numpy

closed 2018-02-17 by ChristianTremblay