ChristianTremblay/BAC0 #55: Raspberry PI 3 -> Python 2.7 -> installing error?

cybermailer, 2017-11-05:
Hi,
i have recently done the following things:
```
sudo apt-get update
sudo pip install bacpypes (installation = no problem)
sudo pip install BAC0 (installation throws errors)
```

here is the screenshot of the error:
[http://aaph.itc-haas.at/resources/jpg/bac0_error.png](url)

Am i doing something wrong?

ChristianTremblay, 2017-11-06:
For my RaspberryPi I used miniconda to install python 3.
I'm not in front of it right now but I guess this should work : 
https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-armv7l.sh

ChristianTremblay, 2017-11-06:
To be honest, I'm not doing any particular effort to be Python 2 compatible... 
https://pythonclock.org/

ChristianTremblay, 2017-11-07:
Just looked to my RPi and I used this : 
https://github.com/jjhelmus/berryconda

closed 2018-01-19 by ChristianTremblay