ChristianTremblay/BAC0 #59: query for point, & set at a specific priority

F4u5t, 2017-11-16:
Hello, I am just getting started with BAC0 and so far i love it,but I need my script to query if a point exists in the controller and then set it at a specific priority, any help?

ChristianTremblay, 2017-11-19:
Hi !
Once the device created using `dev = BAC0.device('address', bacoid, bacnet)` you can use 

dev.points to have a list you can iterate and search for your point name

once found, you can use
```
dev['the_point_I_need'].write(value, priority=3)
```

By default, BAC0 write @ priority 16 on the presentValue (the prop argument of the wrte function)

[ref :  https://github.com/ChristianTremblay/BAC0/blob/ac08eb5c333bbe2868bc5b785d18a8b94a147872/BAC0/core/devices/Points.py#L165)]

F4u5t, 2017-11-19:
Great, thanks! 2 more quick questions, How can I see what priority a point is currently being controlled at and how do I release at a given priority?

ChristianTremblay, 2017-11-20:
This is something we should work on :) 
Actually, in the script, I created 2 properties named "overridden" and "simulated". 
If you simulate an input (setting the out_of_service flag to True and the presentValue to whatever you want) the "simulated" flag will turn on. Same with the overridden if you ever write to priority 8.

But those are set by the app so if you connect to a network, you can't actually tell quickly if some points are set to a certain priority.

To release, we have to send a "null" value to the priority.
BAC0 make this easy in the context of an override or a simulation.

Example with an output : 
```
dev['analogOutput_for_a_valve'] = 100
# This will "override" the output... because it's an output and typically, if you want to write to an output, you want to make an override...
dev['analogOutput_for_a_valve'] = 'auto'
# This will release the override
```
Example with a simulation
```
dev['an_analogInput_temperature_sensor'] = 23
# This will "simulate" the value because if you want to write to a physical analog Input, you probably want to simulate something else...
dev['an_analogInput_temperature_sensor'] = 'auto'
# Revert the simulation
```

F4u5t, 2017-11-20:
The reason i am having trouble is because the point I am playing with is a AV, not a AI or AO
I can write a value, at whatever priority, but there seems to be no way to release it...

***Outside Air Temperature is a a Analog Value
ahu['Outside Air Temperature'].write(7,priority=3) will write the value but when I try to release with
ahu['Outside Air Temperature']="auto"
i get 
ERROR:BAC0.core.devices.Device:Value was not simulated or overridden, cannot release to auto

However if I just directly write and specify a priority - it works
ahu['Outside Air Temperature'].write(7,priority=3)
ahu['Outside Air Temperature'].write('null', priority=3)

the issue i see is that there is no way to tell if a point is being controlled, so you may need to go higher that the default priorities in the scripts - and if you happen to write to the same priority the point is previously being controlled to - you have nothing to release too

ChristianTremblay, 2017-11-24:
I have an idea.... I'll try to keep you updated.

ChristianTremblay, 2018-03-07:
I thought about it and it seems logical to keep the `='auto`attached to an manual operator command (override) @ priority 8.
Other priority should be use with the `.write(value, priority=x)` value being 'null' if we want to release the value.

To me it is a correct behaviour.

Now this probably should be documented correctly.

ChristianTremblay, 2018-09-29:
To see if a point is written to a certain priority, we must read the priority array. It is a specific read operation. 

See 
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Device.py#L335

And

https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Points.py#L120

closed 2021-02-13 by github-actions[bot]