ChristianTremblay/BAC0 #61: readMultiple() empty return

fgervais, 2017-12-20:
We have a pretty abused network, bacnet wise, so I guess I'm running into issues which shouldn't happen to most users.

Today I got this error from one of my script:

> Exception in thread rpm_poll:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/fgervais/.local/lib/python3.5/site-packages/BAC0/tasks/TaskManager.py", line 44, in run
    self.process()
  File "/home/fgervais/.local/lib/python3.5/site-packages/BAC0/tasks/TaskManager.py", line 50, in process
    self.task()
  File "/home/fgervais/.local/lib/python3.5/site-packages/BAC0/tasks/Poll.py", line 71, in task
    self._device.read_multiple(list(self._device.points_name), points_per_request=25)
  File "/home/fgervais/.local/lib/python3.5/site-packages/BAC0/core/devices/mixins/read_mixin.py", line 134, in read_multiple
    points_values = zip(big_request[1][i:i + len(val)], val)
TypeError: object of type 'NoneType' has no len()

It looks like the Read.py:readMultiple() can return None if we expect an ACK but we receive something else. 

```python
if not isinstance(apdu, ReadPropertyMultipleACK):       # expecting an ACK
    log_debug(ReadProperty,"    - not an ack")
    return
```

I don't know about other calls to this function but an empty return will always fail in the case of read_mixin.py:read_multiple().

```python
try:
    request = ('{} {}'.format(self.properties.address, ''.join(request)))
    val = self.properties.network.readMultiple(request)

...

else:
    points_values = zip(big_request[1][i:i + len(val)], val)
    i += len(val)
    for each in points_values:
        each[0]._trend(each[1])
```

Is there a quick fix that would make sense in that case?

Return an empty list maybe?

fgervais, 2017-12-21:
I went with this as a first safe test. Well see how it goes.

```patch
diff --git a/BAC0/core/devices/mixins/read_mixin.py b/BAC0/core/devices/mixins/read_mixin.py
index a6558ac..9464ec6 100644
--- a/BAC0/core/devices/mixins/read_mixin.py
+++ b/BAC0/core/devices/mixins/read_mixin.py
@@ -131,6 +131,9 @@ class ReadPropertyMultiple():
                         raise Exception('Unknown point name : %s' % error)
 
                     else:
+                        if val is None:
+                            print("FG: val is None, workaround")
+                            val = []
                         points_values = zip(big_request[1][i:i + len(val)], val)
                         i += len(val)
                         for each in points_values:
```

ChristianTremblay, 2017-12-22:
I miss a couple hours a day, I will look into this ASAP. Thanks for the proposal.

ChristianTremblay, 2017-12-23:
@fgervais in this particular context, do you have an idea why the anwser is "none" ? Does the device support segmentation ? I'm wondering if we should trig a `SegmentationNotsupported` error...then try to read one property at a time.

See higher in the code, the function will call itself and force the number of point to 1 instead of 25...

fgervais, 2017-12-23:
Well I'm not sure why it did not return an ack at this point.

I'm reading all the device's properties (which is about 80 values) every 30 seconds. The script ran fine for about a week before I got into this case. And it happened only once at this point.

Clearly it's not an error with the code. It really was an error since it's a one in tens of thousands of requests. The problem is just that, if it does happen, the code cannot recover.

If it happens again I will try to catch the error on wireshark to get more insight.

fgervais, 2018-02-19:
Thanks. 

Next time I work on this I'll use the latest from develop and let you know how it goes.

ChristianTremblay, 2018-09-29:
Closing. Reopen if needed.

closed 2018-09-29 by ChristianTremblay