ChristianTremblay/BAC0 #62: Stack Overflow (endless recursion) on DeviceConnected._buildPointList with NoResponseFromController

Itja, 2018-01-19:
First off, thanks for your great library!

With at least one of my devices, I ran into a stack overflow due to an endless recursion in `/core/devices/Device.py`. It begins in `_buildPointList()` of `DeviceConnected`:

        except NoResponseFromController as error:
            self._log.error('Cannot retrieve object list, disconnecting...')
            self.segmentation_supported = False
            self.new_state(DeviceDisconnected)

The call of `new_state()` triggers line

        self.__class__ = newstate
        self._init_state()

which in turn runs `DeviceDisconnected._init_state()`:

    def _init_state(self):
        self.connect()

After a successful connection:

                if segmentation_supported:
                    self.new_state(RPMDeviceConnected)
                else:
                    self.new_state(RPDeviceConnected)

We again are in `new_state` (see above), which then calls `DeviceConnected._init_state()`:

    def _init_state(self):
        self._buildPointList()

...which closes our endless loop, since another `NoResponseFromController` exception triggers the next round.

As this is my second day working with your code, I'll refrain from posting my "solution" as a PR, as it might not be sufficient for not breaking other parts. However, it goes like this:

        except NoResponseFromController as error:
            self._log.error('Cannot retrieve object list, disconnecting...')
            self.segmentation_supported = False
            self._log.info('Changing device state to {}'.format(DeviceDisconnected))
            self.__class__ = DeviceDisconnected

ChristianTremblay, 2018-01-19:
hummm
I will have to think about it.... 

I'm curious about one thing.... does your device implemented a object list ?
you can check that without declaring a device : 
```
bacnet = BAC0.connect()
bacnet.read('2:4 device 4 objectList') #where 4 is BACOID)
```

This would explain why we're stuck there....

Itja, 2018-01-26:
As I scanned a whole network full of devices I don't know which one it was at this point. Most of them work fine, and only some do some strange things under some certain conditions or bad timings. But I'll definitely come back and provide more info here once I meet that one again. Hopefully next week.

ChristianTremblay, 2018-03-22:
@Itja any feedback ?

Itja, 2018-03-28:
Sorry, I never encountered this issue again. Again, thank you very much for your work!

closed 2018-03-28 by Itja