ChristianTremblay/BAC0 #63: History from controllers, Trendlogs

AndrisKrumins, 2018-01-26:
Hello, Christian,

Thank you for opportunity to use BAC0. I am trying to save data points from controllers.
I am trying 2 ways - first default. I recognized, that after some while there is no history from controller.
For example:
cosmos['Merijumi stavos\Z-1 rad. Udems turpgaita'].history
Out[26]: 
2018-01-26 13:02:51.972672    38.799999
2018-01-26 13:02:55.979902    38.799999
2018-01-26 13:03:07.093537    38.900002
2018-01-26 13:03:20.208287    38.799999
2018-01-26 13:03:33.340038    38.799999
2018-01-26 13:03:46.469789    38.799999
2018-01-26 13:03:59.988563    38.700001
2018-01-26 13:04:10.604170    38.700001
2018-01-26 13:04:23.073883    38.700001
2018-01-26 13:04:36.350642    38.700001
2018-01-26 13:04:49.477393    38.700001
2018-01-26 13:05:02.748152    38.700001
2018-01-26 13:05:14.863845    38.700001
2018-01-26 13:05:27.998596    38.600002
2018-01-26 13:05:40.598317    38.600002
2018-01-26 16:21:10.349221    38.900002
Name: COSMOS 810 OPEN/Merijumi stavos\Z-1 rad. Udems turpgaita, dtype: float64

There is no data between last measurement and previous measurement.

Second way to save data was python code, which every minute saves date to sql lite data base. Then after some time I got error message (not equal sizes of arrays):

ValueError: cannot copy sequence with size 1483 to array axis with dimension 148
2

Is there some ways hot to store data from controller for a long period in database? Is it possible to use SQL database built in BAC0?

Thank you,
Andris

ChristianTremblay, 2018-01-29:
Are you sure the computer didn’t fall in sleep mode ? 

Regarding SQL, I’m working hard on the next version actually and i’ve Implemented the possibility to save automatically to a SQLite local database every x poll (default 6 polls...so about 1min). 

I’ve had some comments about the thread sensibility of SQLite but have no better idea at this moment. I’ve also planned a way to clear the history on demand or way data is saved to sql so the user have the flexibility to do what he wanted. 

What is lacking now is a way to append the SQL data to the in-memory dataframe when calling device[‘point’].history 

Have a look to the branch called Bokeh. 

And thanks for your interest !

ChristianTremblay, 2018-03-22:
@AndrisKrumins have you had the chance to look at develop branch and see if problem is still there for you ?

ChristianTremblay, 2018-09-29:
Let’s continue in issue #104

closed 2018-09-29 by ChristianTremblay