ChristianTremblay/BAC0 #64: bacnet.devices returning a Value Error

tybeady, 2018-02-01:
when trying to initialize a bacnet instance I get the following error when calling bacnet.devices:

File "C:\Users\continuum\AppData\Local\Programs\Python\Python35\lib\site-packages\BAC0\scripts\BasicScript.py", line 186, in devices
    deviceName, vendorName = self.readMultiple('%s device %s objectName vendorName' % (device[0], device[1]))
ValueError: not enough values to unpack (expected 2, got 0)

ChristianTremblay, 2018-02-03:
It means that the device do not provide a deviceName or a vendorName (or both).

What kind of controller is it ?

Can you try a direct access to those objects ?
```
address = '2:5' # the device address.....may be an IP addr
deviceID = 1000 # the device ID of the device (BOID)
bacnet.read('%s device %s vendorName' % (address, deviceID))
bacnet.read('%s device %s deviceName' % (address, deviceID))
```

ChristianTremblay, 2018-03-07:
@tybeady can you make a test with this device and the latest develop branch ?

ChristianTremblay, 2018-09-29:
Closing. Reopen if needed

closed 2018-09-29 by ChristianTremblay