ChristianTremblay/BAC0 #65: Can't import BAC0 on Raspberry Pi

candronikos, 2018-02-06:
Using the Python 3 REPL:

```
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170124] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import BAC0
Missing required dependencies ['numpy']
>>>
KeyboardInterrupt
>>> bacnet = BAC0.connect("192.168.0.150/24")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'BAC0' has no attribute 'connect'
```

Doesn't work at all.

More question, is it possible to use this library without bokeh?

ChristianTremblay, 2018-02-08:
Hi ! On my RPi I use miniconda and I've been able to use pandas and bokeh.

That said, your comment is pointing to something I wanted to do for a while which is : having some kind of way to ease the startup of the script without additionnal features like flask, bokeh or pandas.

The next release will support a "lite" version of BAC0 if pandas, bokeh or flask is not installed on the device.

Check out the dev branch in the next couple of days and you'll be able to try.

candronikos, 2018-02-08:
Thankyou, I will give that a try.

I have another issue I encountered when trying to discover the points on a device on my local network. What is the best way to contact you about this?

ChristianTremblay, 2018-02-08:
Right here (creating a specific issue) so we keep track of bugs and problems...
Or here : https://gitter.im/ChristianTremblay/BAC0
where we can chat about the bug

I've just pushed to the develop branch the latest version which is 0.99.902
Have a look

candronikos, 2018-02-12:
Thankyou, the development branch works. Just tested it now and I'm able to read the points

candronikos, 2018-02-12:
I do have one more question. Do you know when these changes can be considered stable?

ChristianTremblay, 2018-02-15:
The develop branch is the branch I use everyday.
That is all I can promise. It's a personnal effort, made during time where children are asleep.
I will welcome any help you can offer

candronikos, 2018-02-15:
Fair enough,

I'm happy to contribute as needed to improve the library as we have an interest in using it. I will have to take some time to understand the architecture though.

closed 2018-02-19 by ChristianTremblay

fgervais, 2018-03-09:
I had the same problem today with pandas complaining about the missing numpy dependency and then the missing connect attribute with BAC0 0.99.100 on the raspberry pi and python3.

Short story:
```
pip3 uninstall numpy
pip3 uninstall pandas
sudo apt-get install python3-pandas
```

Just for reference, I'll post a more verbose explaination.

Basically, the raspberry pi comes with an "old" numpy version (1.12.1) and no pandas. So when we do a pip3 install --user of BAC0 it installs a newer version of numpy and pandas, both of which didn't work in my case.

When launching BAC0 we see a cute little missing dependency message but in fact numpy is crashing like crazy. This can be verified with a 
```
$ python3
>>> __import__("numpy")
```
The solution is to uninstall the new numpy version and stick with the older one.
```
pip3 uninstall numpy
```

Unfortunately, pandas will then complain about its module being compiled for a newer API version of numpy.

To fix this, we simply need to install the raspberry pi version of pandas like so
```
pip3 uninstall pandas
sudo apt-get install python3-pandas
```

ChristianTremblay, 2018-03-10:
@fgervais  What work best for me with Raspberry Pi is a project called berryconda (https://github.com/jjhelmus/berryconda) it installs the Conda environment with a lot of packages for the Pi...and gives access to packages for the Pi. I mean by that... When it is installed, you just have to `conda install pandas` And you will get the latest Pandas version compiled for the Pi.

fgervais, 2018-03-11:
@ChristianTremblay Thanks I'll have a look