ChristianTremblay/BAC0 #68: Point list refresh

fgervais, 2018-02-08:
It's not an issue per se, more of my understanding on how the point list mechanism works that is lacking.

From our recent tests and a quick look inside the code, it doesn't seem like the point list is refreshed, is that correct?

In our application we have a long running BAC0 script that polls different bacnet devices and changes its behavior based on the point list.

How would this script get the updated point list if a bacnet device gets a code update which adds or removes variables?

fgervais, 2018-02-09:
Yes BAC0.dynamic_device(infos) seems a good fit.

I'll give it a go and report back with a pull request.

Thank you

ChristianTremblay, 2018-02-15:
@fgervais I had an idea. I'm asked to add some kind of feature that would detect that a device is offline to be able to correctly handle the reconnection of the device. This would, I think, fit well with your need because we could use what I added recently to the library under the "network" itself : `registered_devices` list. We could iterate this list once in a while to verify that they are still online...

A ping feature. We could choose one information to poll.... could be the device name.... or the object_list length (element 0).... using the latter, we could even detect "some" (not all) modifications of a device objects list.

When a registered device goes offline, we disconnect it. When it comes back we reconnect it and force a refresh of the object list. We'll need to take into account what already exist so we don't overwrite objects histories....

The dynamic device can live inside this pattern as the object list could be changed on the spot...without even loosing the ping (which should not be too short.... like 5 minutes maybe...)

Comments welcomed

fgervais, 2018-02-15:
Yes I think this would fit perfectly.

On the other hand, do you think static devices could also have a function that can be used as a way to manually trigger the refresh of the object list?

I like the idea of having it dynamic but in my application I have idle times where I could trigger the refresh without affecting the rest of the application tasks.

ChristianTremblay, 2018-02-16:
I must think of a way to refresh the list without loosing histories. I don<t want to delete everything and reset the list.
I will need to (maybe) keep track of the last point list. If the device support segmentation... no big deal it’s one request. But if not.... then we must read points one at a time.... 
Maybe a two step process...make the new list..... compare with existing and make adjustments....

If point needs to be deleted....stop any thread attached to it then delete....
New point : create and start polling if necessary

IntoCpp, 2020-01-06:
Any news on this front? 
My two cents: It would be great to offer a manual "device.force_refresh()"; and let users "in the know" refresh a specific device they know they just changed. But still keep the internal dynamic tracking to automate this refresh when a device disappears for a while.

IntoCpp, 2020-01-06:
Also: you stated: "_I'm asked to add some kind of feature that would detect that a device is offline to be able to correctly handle the reconnection of the device._"
Can you elaborate your fix (if you had time to implement) and point me to any documentation on this? 
I'm looking into: "BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : Timeout" that we get on power-reset. (May be related to [issue #60](https://github.com/ChristianTremblay/BAC0/issues/60) ). 
Thanks.

IntoCpp, 2020-01-23:
Quick fix: 
First please note for an easier read: I have a "my_device" object that contains a BACO.device object (with other information and functions for my internal needs).
So my solution: my_device regularly create a **secondary BAC0.device** which will automatically retrieves the list of data-points upon creation. I compare this list with the **current BACO.device** data-point list. If there is a mismatch, I replace the current BAC0.device with the new-up-to-date one and continue.
This works for newly added data_points.
Another solution was needed for deleted data-points. Upon reading a removed point we get an UnknownObjectError exception. Chances are this occurs between 2 check of the first solution. So I now trap this exception (now possible due to pull-request from #60) and renew the connection without having to compare the lists.

Those two tricks can be implemented in a few lines of code, so hopefully it's a good fit for anyone. 
If an internal solution is ever implemented, it would make the client-side code simpler, and somewhat cleaner. I think it would be a nice to have option.

Perhaps @fgervais can comment if this is enough for him so this issue can be closed. I like a clean house :)
 
Here is some pseudo-code from my second trick:
```
from BAC0.core.io import IOExceptions as BAC0_IOExceptions

def test_print_data_point(self, point_index):  
    try:
        print(self.BAC0_device[point_index])
    except BAC0_IOExceptions.UnknownObjectError as e:
        print("Warning: device [{}] data-points out of synch, resetting"
              " BAC0 COM layer to fetch latest data-points: [{}]."
              .format(self.name, e), flush=True)
        # Make sure to save the  'bacnet' object from the first creation.
        self.BAC0_device = BAC0.device(
            self.ip_address, self.device_id, self.bacnet, poll=0)

```
The first trick is a bit longer, but you get the idea.

IntoCpp, 2020-01-29:
I'm facing a big unrelated problem right now, but I will check-out your suggestion when time permits (few days at least).

I can comment right-now for the historical-data: I'm not using that feature, so I would say "kill kill kill". But I read enough comments to understand it is very important for many user.
My suggestion is to keep the data-point, but have it flagged as "unavailable" (how to best do this, I don't know, probably a new bool member in the data-point object). 
The data-point may return later on and we should then resume the historical data, with a jump in time.
If historical-data-space is an issue, than an option could be used to remove or keep it, default to remove.

Have a nice day.

ChristianTremblay, 2020-02-04:
@fgervais @IntoCpp 
If you try the develop branch, I have modified how registered devices are dealt with on the network itself to add a ping feature. It's like a monitoring mechanism that will check if registered devices (defined explicitly with BAC0.device()) are still there. 

If they disappear, BAC0 will disconnect them. This will force a save (by default) and turn them into devices running using the database (so still valid, just no on the network).
If they get back online, BAC0 will attempt a reconnection.

Reconnecting a device makes it recreate the points list so, if new points have been added, they should be there. If points have been removed, they won't be available.

On the develop branch, I kept the ping delay to 10 seconds but it should be higher in real life.

To mimic this if one would need to refresh a device, he could use something like : 
```
device.disconnect(unregister=False)
device.connect(network=bacnet)
```

`register=False` keeps the device in the list so it can be pinged again and in fact, if you disconnect the device and wait... if it can be pinged, it will reconnect.

If you don't need the device anymore you can

```
device.disconnect()
```

by default it uses `unregister=True` and the device gets removed from the list.

Not sure I would want to put a lot of efforts to keep information on removed points... it is miles away from my use case.

@IntoCpp you'll see that it is pretty close to your suggestion. I hope it'll work for you.

IntoCpp, 2020-03-09:
Just switching to the develop branch saw at least a 90% diminution of the "fetch data points" failures.
(some devices will always go AWOL) 

It may be of interest to note that my device connections are initialized using:
`device = BAC0.device(ip, id, bacnet, poll=0)`
It is my understanding that when using '**poll=0**' then no background pooling will occur, until we call "device.points". (Mentioned in case it changes anything under the hood).

Extra: (hijacking this thread topic): I've just started to search how to bring the 'NoResponseFromController' time-out exception down from 10 seconds to 2 seconds. Early detection would give us more options for actions in our short data gathering loop (30 seconds).

A great day to all.

closed 2020-04-08 by ChristianTremblay