ChristianTremblay/BAC0 #69: Bacnet/IP

max13fr, 2018-02-15:
Hello,

I try to read values throught BACNET/IP so I don't have any MSTP or MAC ADDR as requested in BAC0.device() : 

```
# Define a controller (this one is on MSTP #3, MAC addr 4, device ID 5504)
mycontroller = BAC0.device('3:4', 5504, bacnet)
```
Exemple of the whois on my equipments : 

```
pduSource = <Address 192.168.0.10>
iAmDeviceIdentifier = ('device', 1001)
maxAPDULengthAccepted = 1476
segmentationSupported = segmentedBoth
vendorID = 80

pduSource = <Address 192.168.0.11>
iAmDeviceIdentifier = ('device', 1002)
maxAPDULengthAccepted = 1476
segmentationSupported = segmentedBoth
vendorID = 80
```
Can I use BAC0 to poll values on BACNET/IP or it's not planned ?

Thanks in advance,
Best regards, Max

ChristianTremblay, 2018-02-15:
Even when connecting to MSTP devices, BAC0 uses BACnet/IP
```
your_1st_device = BAC0.device('192.168.0.10', 1001, bacnet)
your_2nd_device = BAC0.device('192.168.0.11', 1002, bacnet)
```

closed 2018-02-15 by ChristianTremblay

max13fr, 2018-02-15:
My bad, I tried it but I had a mistmatch between IP and device ID. The error was not really clear : "IndexError: list index out of range."
Work like a charm now.

Thanks,
Max