ChristianTremblay/BAC0 #70: whois() and iam() issues

max13fr, 2018-02-15:
Hello,

I use develop branch on a raspberry pi.

I try to call whois() to get devices list but it didn't return anything :

```
>>> import BAC0
>>> bacnet = BAC0.connect(ip='192.168.0.249')
Starting BAC0 version 0.99.903 (Lite)
>>> bacnet.whois()
defaultdict(<class 'int'>, {})
>>> bacnet.devices
[]
```

According to tshark, network seems ok : 

```
 9723 3694.005977790 192.168.0.249 → 255.255.255.255 BACnet-APDU 54 Unconfirmed-REQ who-Is
 9724 3694.007936326 192.168.0.207 → 192.168.0.255 BACnet-APDU 62 Unconfirmed-REQ i-Am device,1003
 9725 3694.008189346 192.168.0.206 → 192.168.0.255 BACnet-APDU 62 Unconfirmed-REQ i-Am device,1002
 9726 3694.008191065 192.168.0.205 → 192.168.0.255 BACnet-APDU 62 Unconfirmed-REQ i-Am device,1001
 9727 3694.009492155 192.168.0.208 → 192.168.0.255 BACnet-APDU 62 Unconfirmed-REQ i-Am device,1004
 9728 3694.013289852 192.168.0.201 → 192.168.0.255 BACnet-APDU 66 Unconfirmed-REQ i-Am device,77001
 9729 3694.016247604 192.168.0.202 → 192.168.0.255 BACnet-APDU 66 Unconfirmed-REQ i-Am device,77002
```
 
--

When I try to call iam(), I get this error : 

```
>>> bacnet.iam()
Traceback (most recent call last):
  File "/home/max13fr/BAC0/BAC0/core/functions/WhoisIAm.py", line 159, in iam
    iocb.wait()
AttributeError: 'NoneType' object has no attribute 'wait'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/max13fr/BAC0/BAC0/core/functions/WhoisIAm.py", line 163, in iam
    log_exception("exception: %r" % error)
TypeError: log_exception() missing 1 required positional argument: 'txt'
```

Am I missing something ?

Thanks in advance,
Best regards, Max

max13fr, 2018-02-15:
For the issue about whois(), there is large commented part in https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/core/functions/WhoisIAm.py

Is it the source of the problem ?

ChristianTremblay, 2018-02-15:
Try with 
```
bbacnet = BAC0.connect(ip=“192.168.0.249/24”)
# /24 beign your network mask 255.255.255.0
```
Some BACnet router (like Tridium Niagara Jaces) will not respond to global broadcast (255.255.255.255)

max13fr, 2018-02-16:
It works like a charm for whois(), thanks for the fast anwser !

Regarding iam() function, I still have the same issue (but it's less important).

Best regards,
Max

ChristianTremblay, 2018-02-16:
I'll look into the IAM error.... typically thought you should not need to call it manually. BAC0 will answer to iam when asked

closed 2018-02-19 by ChristianTremblay