ChristianTremblay/BAC0 #71: Whois() Issue

spndo, 2018-02-16:
Hi,
We have a python script that is trying to run whois function and get a list of all the devices that responds. But when we try to run it in the terminal, it shows empty dict in the command line. 
However, when we type exact same each line of code in the python shell, it works and gives the list of responding devices. We try to debug but still get nothing. 
Is it possible that you can give us advice on how to solve this? (Wireshark works well when we run the script)
`import BAC0

bacnet = BAC0.connect('10.20.0.169/22', bokeh_server=False)

who = bacnet.whois()
print(who)
`
Thank you so much!

[image]
[image]`

ChristianTremblay, 2018-02-17:
When you work in the shell, there is a delay between the moment where you send the Whois request and the moment where you type `print(who)`. 
This delay do not exist when you call the script itself, you must give time for the request to be processed and an answer to be obtained.
To be clearer, `who` exists as soon as you declare it. So `print(who)` can run as soo as the request is sent... no need to wait for an answer.

BACnet messages are processed in a thread so you have to be careful with those timing issues.

Now, depending on what you need to do with your script, you could use an “inelegant but working” workaround like 
```
import time
import BAC0
bacnet = BAC0.connect()
bacnet.whois()
time.sleep(2)
print(bacnet.devices)
```

spndo, 2018-02-21:
Hi Christian,

The code works well now. Thank you so much for the help!

closed 2018-02-21 by spndo