ChristianTremblay/BAC0 #74: Getting Exception When connect

spadilam, 2018-03-12:
Team,
I am facing the below issue with respect to connection to BACnet device.
>>> import BAC0
>>> bac = BAC0.connect(bokeh_server=False)
Starting BAC0 version 0.99.105
Using ip : 169.254.1.2
WARNING:BAC0.script.ReadWriteScript:Bokeh server not started. Trend feature will
 not work
>>> Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\test\AppData\Local\Programs\Python\Python36\lib\threading.py
", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\test\AppData\Local\Programs\Python\Python36\lib\threading.py
", line 864, in run
    self._target(*self._args, **self._kwargs)
TypeError: run() got an unexpected keyword argument 'sigterm'

ChristianTremblay, 2018-03-12:
What version of bacpypes are you using ?
Also, have a look to develop branch to try BAC0.lite feature that will prevent the web app and bokeh from being used

spadilam, 2018-03-13:
Currently using bacpypes-0.15.0 version

spadilam, 2018-03-13:
Installed BAC0.lite from development branch.
getting the below errror

>>> import BAC0
>>> BAC0.connect()
Starting BAC0 version 0.99.909 (Lite)
Traceback (most recent call last):
  File "C:\Users\padilas\AppData\Local\Programs\Python\Python36\lib\site-package
s\BAC0\core\functions\GetIPAddr.py", line 59, in _findIPAddr
    s.connect(('google.com', 0))
socket.gaierror: [Errno 11001] getaddrinfo failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\padilas\AppData\Local\Programs\Python\Python36\lib\site-package
s\BAC0\scripts\Lite.py", line 71, in __init__
    host = HostIP()
  File "C:\Users\padilas\AppData\Local\Programs\Python\Python36\lib\site-package
s\BAC0\core\functions\GetIPAddr.py", line 30, in __init__
    ip = self._findIPAddr()
  File "C:\Users\padilas\AppData\Local\Programs\Python\Python36\lib\site-package
s\BAC0\core\functions\GetIPAddr.py", line 65, in _findIPAddr
    'Impossible to retrieve IP, please provide one manually')
BAC0.core.io.IOExceptions.NetworkInterfaceException: Impossible to retrieve IP,
please provide one manually

ChristianTremblay, 2018-03-13:
You need to provide an IP

BAC0.connect(ip=‘the IP of the interface you are using / mask bit’)

Not sure but I think you should upgrade bacpypes to at least 0.16

ChristianTremblay, 2018-03-22:
@spadilam have you been able to make it work ?

spadilam, 2018-03-22:
@ChristianTremblay  Tried with bacpypes 0.16 and able to connect to device was able to read the parameter thanks for the support

closed 2018-03-22 by spadilam