ChristianTremblay/BAC0 #75: Improve error "IndexError: list index out of range"

max13fr, 2018-03-12:
Hello,

When you try to get a point that doesn't exist on a device during a read_multiple (with segmentation), the exception is not helping : "IndexError: list index out of range" (i use the develop branch).

For example : 

```
import BAC0;
bacnet = BAC0.connect(ip='192.168.1.1/24');
obj_list = [('analogValue', 31), ('analogInput', 40)];
device = BAC0.device('192.168.1.12', 1002, bacnet, object_list = obj_list)
```
```
Starting BAC0 version 0.99.904 (Lite)
 ! <bacpypes.basetypes.ErrorType object at 0x761bd0d0>
 ! <bacpypes.basetypes.ErrorType object at 0x761bd370>
 ! <bacpypes.basetypes.ErrorType object at 0x761bddb0>
 ! <bacpypes.basetypes.ErrorType object at 0x761bde10>
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 154, in __init__
    self.new_state(DeviceDisconnected)
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 165, in new_state
    self._init_state()
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 599, in _init_state
    self.connect()
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 625, in connect
    self.new_state(RPMDeviceConnected)
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 165, in new_state
    self._init_state()
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 343, in _init_state
    self._buildPointList()
  File "/home/max13fr/BAC0/BAC0/core/devices/Device.py", line 407, in _buildPointList
    self.properties.objects_list, self.points = self._discoverPoints(self.custom_object_list)
  File "/home/max13fr/BAC0/BAC0/core/devices/mixins/read_mixin.py", line 217, in _discoverPoints
    point_infos = analog_points_info[i]
IndexError: list index out of range
```

I found the source of the issue, it's in BAC0/core/devices/mixins/read_mixin.py : 

We send a request on 2 points (analogValue 31 objectName presentValue units description analogInput 40 objectName presentValue units description) but we only get one reply.
As a list as used to store points, there is a inconsistency between the request points list and the values.

The exception "IndexError: list index out of range" is a side effect of this issue.

This behaviour is really unsafe because there is a risk about a shift of the values on the points.
For example, if we try to get addresses 1 & 30 & 5 but with 30 not replying anything, the points list will be invalid : 
- index 0 : containing point #1 => ok
- index 1 : containing point #5 instead of None value of #30 
- no index 2

In my opinion, the best fix is to check (type & address) in the result of read_multiple() to add None value if point not found (line 118) : 

```
                    else:
                        for points_info in self._batches(val, info_length):
                            values.append(points_info)
```

I didn't send any pull request to fix that because I don't know the repercution.

I hope my explanation was clear.
Thanks for your fabulous work on this module

Best regards, Max

ChristianTremblay, 2018-03-12:
Good catch. I will have to look. Feel free to make a PR. It will at least give a kind of start

ChristianTremblay, 2018-03-15:
This is in fact really more complicated that I thought it was and the problem occurs only if we provide a wrong object list when defining a device. 

Continuing the tests...

ChristianTremblay, 2018-03-22:
@max13fr Please have a look to develop branch where I added a `UnknownObjectError` that handle this case.

ChristianTremblay, 2018-04-05:
Fixed in develop

closed 2018-04-05 by ChristianTremblay