ChristianTremblay/BAC0 #76: How to enable log_debug() in the console?

fgervais, 2018-03-13:
I'm on the develop branch and I'd like to enable the log_debug() messages in Read.py.

I've been bruteforcing for a solution trying:

_DEBUG = 1
_debug = 1
_LOG = ModuleLogger(globals())
_log = ModuleLogger(globals())

I don't really get how the debug work so I don't know what I'm aiming for.

Any idea?

ChristianTremblay, 2018-03-13:
Yeah sorry I broke something recently...
The logging feature has always been clunky as I was using bacpypes logging ... mixed with BAC0 things.... Now i’m trying To use core.utils.note decorator but I didn’t manage to make it work with Read.py

My goal is to be able to include logging information in the web app. A lot of trials and errors as I learn.

fgervais, 2018-03-13:
Let me know what you think of this:
https://github.com/fgervais/BAC0/commit/5912b1fff24b44fd4300b3fa8013117a1adf77ce

I'm not sure if that's what you had in mind.

Also I used _log.error() to replace log_exception(). You might want to handle this differently.

ChristianTremblay, 2018-03-15:
Can you test new version of develop ?
It is mostly what you provided with some little modification..... I’m also slowly improving error handling with a new ‘unknowObject’ error... 
To be continued as I plan to replace self._log calls with a function that will allow us to enable or disable logging.... for now, it is on by default under home/.BAC0/BAC0.log and in the console.

fgervais, 2018-03-15:
Yes I'll test it out and let you know

fgervais, 2018-03-15:
I get at least 1 failure.

Changing from '%' to ',' fixes the problem somehow.

```
Traceback (most recent call last):
  File "log_server.py", line 65, in <module>
    test_dev = BAC0.device('172.16.11.55', 11055, bacnet)
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 154, in __init__
    self.new_state(DeviceDisconnected)
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 165, in new_state
    self._init_state()
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 599, in _init_state
    self.connect()
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 625, in connect
    self.new_state(RPMDeviceConnected)
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 165, in new_state
    self._init_state()
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 343, in _init_state
    self._buildPointList()            
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/Device.py", line 407, in _buildPointList
    self.properties.objects_list, self.points = self._discoverPoints(self.custom_object_list)
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/mixins/read_mixin.py", line 213, in _discoverPoints
    analog_points_info = self.read_multiple('', discover_request=(analog_request, 4), points_per_request=5)
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/devices/mixins/read_mixin.py", line 98, in read_multiple
    val = self.properties.network.readMultiple(request)
  File "/media/fgervais/c0b0db62-ee5f-4b1b-84fb-c303406e5b57/BAC0/BAC0/core/io/Read.py", line 213, in readMultiple
    objectIdentifier)
TypeError: not all arguments converted during string formatting
```

ChristianTremblay, 2018-03-21:
Ch't'embêté

ChristianTremblay, 2018-03-21:
I changed all old string format to {}.format as you saw... really not sure if the error came from there but I took that as an excuse to uniformize the code... 
Is your PR linked to this bug in some way ?

fgervais, 2018-03-21:
No the PR is not linked to this but your work changing to .format() fixed this issue.

In other words, with the current develop branch state, this issue is solved.

ChristianTremblay, 2018-03-22:
@fgervais Closed that but there was still something missing which is now in the develop branch

```
BAC0.log_level(logging.INFO)
```
will dynamically change the logging level
You will also see that I made a lot of modification to the rendering of Read.py....other will follow
Now the information is much clearer.

fgervais, 2018-03-22:
Hehe yes I saw that but a bit too late unfortunately.

Only once I had the patch ready I updated to test with the latest version and I saw the much better indented debug info version.

ChristianTremblay, 2018-04-05:
Fixed in develop

closed 2018-04-05 by ChristianTremblay