ChristianTremblay/BAC0 #78: High memory usage

fgervais, 2018-03-22:
I've been running a small app that just reads about 500 points every 30 seconds using the develop branch.

After 6 days, the app was using close to 500 megs of ram and it looks like it would keep on increasing.

Any idea where this could come from?

ChristianTremblay, 2018-03-22:
By default, each reading will be added to a list (kept in-memory).
6days, 500 point = 8640000 records

There is a auto-save feature (to sqlite)
https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/core/devices/Device.py#L102

You can clear histories manually : https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/core/devices/Device.py#L275

https://github.com/ChristianTremblay/BAC0/blob/develop/BAC0/sql/sql.py#L82

I have not used that intensively.... it's really a work in progress. It may be useful for you.

One goal I have is to be able to access old data (in sqlite) without effort when there are available data... actually, you will have to dig into the sqlite db file.

fgervais, 2018-03-22:
Thank you, I have a look

ChristianTremblay, 2018-04-05:
Any improvement ?

fgervais, 2018-04-05:
Well.. yes but for the moment I just completely disabled point trends.

```
diff --git a/BAC0/core/devices/Points.py b/BAC0/core/devices/Points.py
index fdf8055..321f5ea 100644
--- a/BAC0/core/devices/Points.py
+++ b/BAC0/core/devices/Points.py
@@ -104,7 +104,7 @@ class Point():
         try:
             res = self.properties.device.properties.network.read('{} {} {} presentValue'.format(
                      self.properties.device.properties.address, self.properties.type, str(self.properties.address)))
-            self._trend(res)
+            #self._trend(res)
         except Exception:
             raise Exception('Problem reading : %s' % self.properties.name)  
```

It's now been running for 13 days and the memory usage seems to be stable.

For now this will have to do but when I get time I'll try to add a parameter to disable trends when instantiating a device or something.

I get that my use-case might not be what you have in mind for your library. You can close the issue if you want.

ChristianTremblay, 2018-04-06:
I see.... you are probably using the library in a larger one that deal with the result...
In that case, you probably don't need polling ?
Can you share your use case ?

fgervais, 2018-04-20:
Yes it's really simple really.

I'm just transparently reading all the bacnet points of multiple devices on a network, as they run their control loop, and store the values on a server.

ChristianTremblay, 2018-04-21:
If you do all the work on your script side, setting poll=0 would probably help....
Or I could maybe plan to add some kind of "roll over feature" for trend.... if a value is provided, I could limit the number of records in memory instead of growing to infinite....and beyond !
ex. `history_size=500`
Your script would then let the polling do its job....and you read the `devicec['point'].lastValue`  read (instead of using `devicec['point'].value` on a point which force a network read...

In fact, I like the `history_size` feature.....it brings flexibility... will make some tests... could be 1 minimum and last value is always available....

Thanks for sharing

fgervais, 2018-04-23:
Yes that would be perfect

ChristianTremblay, 2018-04-24:
@fgervais  Have a look to develop branch when you have time....

```
dev = BAC0.device('2:4',4,bacnet,history_size=2)
# or after...
dev.update_history_size(100)
# or just on one point : 
dev['point'].properties.history_size = 30
```

Now I must decide if I make default value something else than `None` ..... What do you think ?

fgervais, 2018-04-24:
@ChristianTremblay that was fast. I'll try it in the upcoming weeks when I need to restart the script.

Can you let me know what is the internal polling rate?

As for the default what is the effect of None? This value will affect the number of points of the graphs in the webapp right?

I think I would go for some relatively small value that doesn't end up using too much ram but still makes interesting graphs out of the box.

At first glance, 100 seems a comfortable number.

ChristianTremblay, 2018-04-24:
@fgervais 
I'm actually working on a really big project where I use BAC0 a lot to troubleshoot network issues so I'm building some features when I feel something would be useful.... like : 

`bacnet.find_devices_on_network(network_number)`
a simple whois with a filter
or
`device.find_overrides()` which will verify if something is written @ priority 1 or 8... a common case in startup, someone just forgot an override and the Niagara station didn't see it...

That's why I'm working on BAC0 a lot those days. Slicing the history was a good idea and I can see the usage to limit memory issues.
The default poll rate for a device is 10sec which is set low because the first goal for BAC0 was to test the control sequences I programmed.... and as I don't have COV yet.... thought that 10sec would be perfect. For any other situation involving long term, or a lot of devices... it would not be a good idea to leave this value so low... BAC0 is made to read every points of a device at each poll rate.... can be really high demand .... "surtout quand y'a du Carel d'impliqué........" with their controllers with thousands of points...

For history_size, the effect of `None` is : no limit.... the list will add records forever. When running the script for short period of time, it's no issue.

100.....or 1 ! The script will assert that the value is at least 1....so the last read value is always available using device['point'].lastValue (which will read the history without generating a read request).

closed 2018-06-09 by ChristianTremblay