ChristianTremblay/BAC0 #79: Segmentation problem

pisuke, 2018-03-30:
Hi, I am trying to use BAC0 to connect to a BACnet device, however I seem to bump into a segmentation problem.
I've tried to disable segmentation with the following command:

controller = BAC0.device('10.8.253.51', 391, bacnet, segmentation_supported=False)

In the console I get a few hundred lines of  this error
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...
ERROR:BAC0.core.devices.Device:Cannot retrieve object list, disconnecting...

and in Wireshark I get this:

1141314	6508.889930	10.8.253.51	10.8.253.50	BACnet-APDU	60	Abort           segmentation-not-supported[ 39] 

Any clue?

Thanks!
Francesco

ChristianTremblay, 2018-04-03:
Hi ! Can you try the develop branch ? There has been a lot of work done over that kind of problem.

pisuke, 2018-04-13:
Sorry, please disregard my previous message. I've upgraded bacpypes and after restarting the jupyter kernel all the functions are back.

pisuke, 2018-04-14:
Hi again, now the instantiation of the controller object starts exchanging information with the BACnet device. 
The device has about 18000 points (it's a gateway) and it takes a while to process, but after a few minutes of processing it stops with this error:

UnknownPropertyError: Unknown property 10.8.253.51 multiStateInput 197564 stateText 

It looks like the stateText property has not been implemented.
I'll comment out the exception and see how it goes for the moment.
I'll do more code reading in the meanwhile.
 
Is this the right document to learn how to use and contribute to the project?
https://media.readthedocs.org/pdf/bac0/latest/bac0.pdf

Thanks!

ChristianTremblay, 2018-04-15:
18000 points ! issshhh.... that is a lot.
I strongly suggest you work with `poll=0` when defining the device...... or another value, much higher.... you won't be able to poll 18000 objects every 10 seconds.
And also, if possible, try to `use custom_object_list`
see : http://bac0.readthedocs.io/en/latest/controller.html

It would allow you to create multiple devices reading different points from this gateway. A few points at a time.

pisuke, 2018-04-15:
Thanks Christian, indeed there are too many points!
The custom_object_list option sounds good, I'll give it a go.
By commenting the exception in the source code I also managed to load all the points.
Thanks for the good suggestions.

closed 2018-09-29 by ChristianTremblay