ChristianTremblay/BAC0 #82: Invalid property for object type getting when trying to access Device properties

dayanand4u, 2018-04-16:
I am trying to access few Device object proprietary properties....but its showing below error 

bacnet.read('192.168.168.6 device 240006 vendorname'))

**ValueError: invalid property for object type**

I want to know the supporting property list of device object like location ,vendor name,vendor id,model name,firmware-revision ....

Could you please help me to resolve this.

ChristianTremblay, 2018-04-16:
bacnet,read('192.168.68.6 device 240006 vendorName')
little typo

You can refer to bacpypes doc for more details about the different properties for each objects
https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py

dayanand4u, 2018-05-17:
Thanks for support. I am able to find the property names of bacnet objects in the mentioned file and execution fine.

I have one more Query ,my module Firmware supports custom objects and few new custom properties to device object ... how to access the device object proprietary property id....

Ex:  Object          ---- Device
      Property ID   ---- 712
I want to read/write the custom objects and properties .Please help me ...

ChristianTremblay, 2018-05-18:
Proprietary objects are a little more complicated. I've started to play with them using the TEC3000 from Johnson Controls. You will have to define the property before being able to define it. Then you will have to register it.
Then you will need to use direct access from bacnet.read/write as I have not yet integrated a way to add it to the device.

Have a look here : 
https://bac0.readthedocs.io/en/latest/proprietary_objects.html
and here
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/proprietary_objects/jci.py

dayanand4u, 2018-05-23:
Could you please help me out how to define and register new property id for device object and
how to define to new proprietary  object and property id and register it.

dayanand4u, 2018-05-23:
Configured new property ID BAC0/core/proprietary_objects/jci.py like below.
Here Object - device and property id- 712

class TECSupOnline(DeviceObject):
    objectType = 'device'
    uiinputs = 712

    properties = [
        Property(712, Boolean, mutable=True),
        ]

    def __init__(self, **kwargs):
        if _debug: TECSupOnline._debug("__init__ %r", kwargs)
        DeviceObject.__init__(self, **kwargs)

def register(cls, uiinputs=712):
    register_object_type(cls, uiinputs=uiinputs)

then added   row   as  ReadableProperty('uiinputs', Unsigned) in bacpypes\device.py file under Device object ...

From script bacnet.read('2401:4 device 4 uiinputs') Its returning always like  invalid property for object type

Please help me out ....

ChristianTremblay, 2018-05-23:
As shown here, `vendor_id` is mandatory
https://bac0.readthedocs.io/en/latest/proprietary_objects.html

For more details, please refer to : 
https://github.com/JoelBender/bacpypes/blob/master/samples/VendorAVObject.py
and
https://github.com/JoelBender/bacpypes/blob/master/samples/VendorReadWriteProperty.py

As explained, proprietary objects are not simple. Lots of details to figure out.

luli440, 2020-07-03:
> 我正在尝试访问一些设备对象专有属性....但其显示如下错误
> 
> bacnet.read（'192.168.168.6设备240006供应商名称'））
> 
> **ValueError：对象类型的属性无效**
> 
> 我想知道设备对象的支持属性列表，例如位置，供应商名称，供应商ID，型号名称，固件修订版...。
> 
> 您能帮我解决这个问题吗？

Vendorname, where to get the value of, please help me

ChristianTremblay, 2021-02-05:
http://www.bacnet.org/VendorID/BACnet%20Vendor%20IDs.htm

closed 2021-02-13 by github-actions[bot]