ChristianTremblay/BAC0 #83: BACnet stack not running - use startApp()

mosterme1003, 2018-05-18:
Hello,

over the last weeks I received following error from time to time:

> File "D:\Regleradaption\Regleradaption\venv\lib\site-packages\BAC0\core\io\Write.py", line 80, in write raise ApplicationNotStarted('BACnet stack not running - use startApp()') BAC0.core.io.IOExceptions.ApplicationNotStarted: BACnet stack not running - use startApp()

Since last week I'm not able to run my script anymore without this issue. I tried to use the startApp() in my code but nothing has changed (probably not properly since I'm quite new to programming). How should I proceed? Thank you for your help, here is my code.

`import time
import matplotlib.pyplot as plt
from BAC0.scripts.BasicScript import BasicScript
from BAC0.core.io.Read import ReadProperty
from BAC0.core.io.Write import WriteProperty

class ReadWriteScript(BasicScript, ReadProperty, WriteProperty):
    pass

class plot:

    @staticmethod
    def readWriteStep(device_address, bacnet):

        time_passed = []
        seconds = 0
        y_output = []
        x_output = []

        bacnet.write(device_address + ' analogInput 6015 presentValue 30')
        bacnet.write(device_address + ' analogOutput 7006 presentValue 30')

        for i in range(30):
            y_present_value = bacnet.read(device_address + ' analogOutput 7006 presentValue')
            x_present_value = bacnet.read(device_address + ' analogInput 6015 presentValue')
            y_output.append(y_present_value)
            x_output.append(x_present_value)
            if seconds == 20:
                bacnet.write(device_address + ' analogInput 6015 presentValue 50')
                bacnet.write(device_address + ' analogOutput 7006 presentValue 90')
            seconds += 1
            time_passed.append(seconds)
            time.sleep(1)
            
        plt.figure(1)
        plt.subplot(211)
        plt.plot(time_passed, x_output)
        plt.axis([0, 60, 0, 100])
        plt.subplot(212)
        plt.plot(time_passed, y_output)
        plt.axis([0, 60, 0, 100])
        plt.show()

class Application():

    @staticmethod
    def main():
        device_address = '139.6.154.83'
        bacnet = ReadWriteScript(localIPAddr='139.6.154.71')
        plot.readWriteStep(device_address,bacnet)

if __name__ == '__main__':
    Application.main()`

ChristianTremblay, 2018-05-21:
import BAC0
import time
import matplotlib.pyplot as plt

class plot:
    @staticmethod
    def readWriteStep(device):
        device['TempA'] = 30
        device['CoilA'] = 30
        
        for i in range(30):
            if i == 20:
                device['TempA'] = 50
                device['CoilA'] = 90
            else:
                # just read it so it will be recorded
                device['TempA']
                device['CoilA']
            time.sleep(1)
            # Note here that thread nature of BAC0 and the speed of the BACnet network 
            # will maybe create imperfect 1 seconds interval readings
        
        plt.figure(1)
        plt.subplot(211)
        plt.plot(device['nciPIDHumTI'].history)
        plt.subplot(212)
        plt.plot(device['nciPIDDeshumTI'].history)
        plt.show()
        device['TempA'] = 'auto' 
        # release the override of the input as it is not typical to write to the presentValue of a analogInput
        print(device['TempA'].history)
        # Just for fun, print the history

class Application():
    @staticmethod
    def main():
        device_address = '139.6.154.83'
        device_instance = 7000 # presuming device instance is 7000.... pick yours
        object_list = [('analogInput',6015), ('analogOutput',7006)]
        bacnet = BAC0.connect() # presuming 255.255.255.0 mask
        device = BAC0.device(device_address,device_instance,bacnet, object_list = object_list, poll=0)
        # Here I will presume I know the point names... let say AI:6014 is called TempA
        # and AO:7006 is named CoilA
        # poll=0 will disable poll... the value will be saved when we read or write them

        plot.readWriteStep(device)
        bacnet.disconnect()
```

This is probably not a good idea to read a lot of points every seconds.... and there could be better ways to do that... just wanted to illustrate how I would use BAC0 features to do what you want

Another way would be using the embedded Web Server / Bokeh Feature and use
```
device['TempA'].chart()
device['CoilA'].chart()
```
And check the live trends on your browser

Thanks for your interest in BAC0 !

regostar, 2018-07-18:
bacnet.disconnect() was not present thats the main reason why you get this error

Safrone, 2018-07-18:
Yeah, this is most common when you have a bacnet instance running somewhere since there cannot be multiple bacnet networks active on one computer as it reserves the IP