ChristianTremblay/BAC0 #84: Automatic batching for SegmentationNotSupported devices

Safrone, 2018-05-21:
I have a bacnet device that does not support segmentation, currently all information about each point under the device is read one by one. For a device with 477 points, this can take awhile.

Is there any plans to automatically batch the requests with readPropertyMultiple? I'm not sure how to calculate the proper batch size but it seems like it would be fairly simple to implement after doing that.

Let me know if this makes sense

Safrone, 2018-05-21:
I'm a bit confused by `RPDeviceConnected` and `RPMDeviceConnected`. Currently the device is set to use `RPDeviceConnected` if segmentation is not supported but especially in that case, one should use read multiple with batching to avoid too many queries.

ChristianTremblay, 2018-05-22:
You are right about batching. BAC0 do that in some way but is not perfect. Don’t hesitate to send a PR if you have an idea. 

Devices with lots of points ( mostly Carel...) often don’t support segmentation. Go figure... 

my preferred way to deal with them is to use a custom object_list when i define the device. It is more than probable that you do not need all 477 variables....

ChristianTremblay, 2018-05-24:
I must admit it's a "panier de crabes" this problem. There are a lot of bad implementations on the field and regarding MSTP devices.... it's really easy to reach to max length.

Maybe a new parameter added to the device that would "override" the RP behaviour. 

`force_rpm = True`

and probably some kind of fine tuning for `points_per_request` (default = 25)

This should be done in BAC0/core/devices/mixins/read_mixin.py

Line 125 (the part where we read points.... not the part where we build the point list)
You could try something like : 

```
if self.properties.force_points_per_request:
    points_per_request = self.properties.force_points_per_request
```

It will require fine tuning... I agree that I've been pretty extreme passing from 25 to 1.... but I prioritized a working BAC0 over a fast BAC0 :-)