ChristianTremblay/BAC0 #85: Regarding  read multiple & write multiple property

dayanand4u, 2018-05-22:
Please provide me some examples for read multiple & write multiple property and also

need some inputs for how to register new property ID name or number for device object and how to register new custom objects & custom property ID' .

dayanand4u, 2018-05-23:
I am able to perform  readmultiple property but i am not find write multiple property in write.py
Could you please guide me to perform writemultiple property service.

Also Please sujjest for below Query:
how to register new property ID name or number for device object and how to register new custom objects & custom property ID' provide some examples..

ChristianTremblay, 2018-05-23:
Write multiple is not implemented and not planned yet.

ChristianTremblay, 2018-05-24:
What is your use case ? Just curious here....  Time is not on my side those days though... I don't want to make promises I won't hold.

dayanand4u, 2018-05-24:
My controller supports custom object Universal inputs(12) and Universal outputs(8) and these custom objects have custom property Id's.so here my use case is to write few property Id 's for all input channels (12) or output channels(8) using write_property_multiple service at a time.If possible please kindly give support.

ChristianTremblay, 2018-09-29:
@dayanand4u please have a look to this regarding proprietary objects

https://bac0.readthedocs.io/en/latest/proprietary_objects.html

Write multiple is still not implemented though. Sorry.

ChristianTremblay, 2020-01-29:
Write Multiple is now implemented in BAC0 (see develop branch)

closed 2020-01-29 by ChristianTremblay

shashankgowdasd, 2023-09-19:
I am unable to use the writeMultiple property and getting an error as **unrecognizedService**, So please help me to find out what changes I need to make for the below code to access it

r = ['analogValue 0 presentValue' ,'analogValue 1 presentValue']
bacnet.writeMultiple(addr='192.XXX.X.XXX',args=r,vendor_id='842')

ChristianTremblay, 2023-09-19:
Unrecognized service is sent by the device you try to write to. It tells you that it doesn't support that feature.