ChristianTremblay/BAC0 #86: Get Point from Device via object_list identifier

Safrone, 2018-05-22:
There does not seem to be a convenient way to get a `Point` from a `Device` using the value that is passed into the `object_list` parameter e.g.:

It could work something like this: 
```
>>> device(..., object_list=[('analogInput', 1), ('analogInput', 2)])
>>> device.get_object(('analogInput', 1))
NumericPoint
```

ChristianTremblay, 2018-05-22:
This is by design as I was seeing no advantage to do so. If you call `device.points` you will get a point list by name, which is IMHO much more convenient. Why would I use IP address when I can use DNS...

device['point_name'] will give you all features from comparison, get, set.... properties are there also (`device['point_name'].properties`) and you can even get a pandas series for histories using `device['point_name'].history

If you still need this, you can probably build something around iterating devices.points and look into properties.

Safrone, 2018-05-23:
Ultimately I built my own reference like so: (Could be made better by mapping the properties instead of querying twice)

```
reference = {
    (device[name].properties.type, int(device[name].properties.address)): name 
        for name in device.point_names
}
```
My thoughts on this is that you basically already have to know the addresses of the points (and ostensibly what they relate to) if you want to use `object_list` in the first place. So now you need to store both of those values to first set up the device and then get the point later.

As to IP address vs DNS, why limit to only DNS?

Finally, I'm not sure how strict the bacnet protocol is on having duplicate point names (obviously this is discouraged) but if it did happen, it would be impossible to query them specifically.

Small note is that Device currently iterates over all points to find the matching name on any query. Not the hugest issue but I feel it could build a reference dictionary from name -> point on initialization. 

Thanks for working on this project, it has made dealing with bacnet much more understandable and consistent.

ChristianTremblay, 2018-05-24:
I should probably start from scratch explaining why I initially built BAC0... I needed a way to simulate/validate variables on controllers I'm programming so I could test the sequences of operation. I could say that when I started this project I had really minimal experience in Python so yes, it is filled with weird things that I hopefully, improve over time. Now that I see the power it gives me, I use BAC0 more and more on job sites when I'm starting up jobs... And it brings new use cases that I build when I need them.

That said, I'm also building the tool mirroring the way I understand the way controllers work and typically, controllers are well made and provide "reasonable" number of variables and are typically providing basic properties.

The custom object_list became a need when I started to interact with Carel controllers... thousand of points, lack of properties... Not cool at all. 

More recently, I played with the Johnson Controls TEC3000 and fought with a multistate value conteining over 200 state text... with no support for segmentation.... on a variable I didn't need to make my tests.

Knowing that the object list can be retrieved easily with a request, I simply used the same format and I assumed that once the device will have been created, I'll just use point names. 

And technically, if I only want to write or read a value without the need to know its name... I will not bother create a device and just use `bacnet.read('address property number presentValue')` ...

Now don't get me wrong, I'm not closed to the idea of bringing another way to access a point.... 

`device[('analogInput',1)]` would probably be possible/useful

> My thoughts on this is that you basically already have to know the addresses of the points (and ostensibly what they relate to) if you want to use object_list in the first place. So now you need to store both of those values to first set up the device and then get the point later.

To that, it's probably related to the workflow of the user. Usually, when I use the custom object_list, I add only the few points I need for the work I do.

> As to IP address vs DNS, why limit to only DNS?

Door isn't closed

> Finally, I'm not sure how strict the bacnet protocol is on having duplicate point names (obviously this is discouraged) but if it did happen, it would be impossible to query them specifically.

Here you have a point. But as you point, we can think it would be a really bad implementation... one could get over this using a custom_object list excluding one of the point

> Small note is that Device currently iterates over all points to find the matching name on any query. Not the hugest issue but I feel it could build a reference dictionary from name -> point on initialization.

You get a point here... but I'm curious to know what we would gain ? Working with MSTP devices most of the time, accessing the points in the device is not my biggest bottleneck :-) And I used a generator to mitigate the access time/memory usage.

> Thanks for working on this project, it has made dealing with bacnet much more understandable and consistent.

Let say I learn everyday building and using it. Thanks for your interest and don't hesitate to provide a PR if you think you can improve the tool.

Safrone, 2018-06-11:
Yeah I think we have different requirements from the framework as a whole. 

I'm using this in an tiny computer to get data from bacnet systems and push it to the cloud. For me, bacpypes took too much effort without getting very much out of it and this project was much closer to a plug and play api wrapper that is basically all I really want.

using `bacnet.readMultiple("...")` after building my desired mapping (and maybe some fallback code in case segmentation is an issue) is probably more sustainable since I'm handling the data storage aspects anyway

ChristianTremblay, 2018-06-12:
Absolutely ! The result is what matters at the end. Don't hesitate to make a PR if you find something helpful in your development. Or maybe some kind of "Example" as a documentation file that could be added to readthedocs.

I'm pretty sure your use case could be interesting for someone else.

Safrone, 2018-06-12:
This is probably the most useful thing I made feel free to add/edit if you want:

used to get the presentValue from a number of points.

```
class BACnetDeviceClient:
    """Device wrapper on top of the BAC0 network"""
    def __init__(self, network, address, object_list):
        """
        Args:
          network: BAC0.connect()
          address (str): address of the bacnet device desired
          object_list (list): List[Tuple[object_type, object_address]]
                              e.g. [('analogInput', 1), ('analogInput', 3)]
        """
        self.network = network
        self.object_list = object_list
        request = " ".join(["%s %s presentValue" % (object_type, object_address)
                            for object_type, object_address in object_list])

        self.request = "%s %s" % (address, request)
        self.network.register_device(self)

    def get_data(self):
        return self.network.readMultiple(self.request)

    def disconnect(self):
        self.network.unregister_device(self)

    @staticmethod
    def get_device_address(device_id, network):
        address, _device_id = next(
            (k for k in network.whois(str(device_id)).keys()
             if (isinstance(k, tuple)) and k[1] == device_id), (None, None)
        )
        if address is None:
            raise ValueError('No device found with device_id=%s on network: %s'
                             % (device_id, network,))
        return address
```

```
import BAC0
bacnet = BAC0.connect()
device_id = 12345
address = BACnetDeviceClient.get_device_address(device_id, bacnet)
object_list = [('analogInput', 1), ('analogInput', 3)]
client = BACnetDeviceClient(network=bacnet, address=address, object_list=object_list)

client.get_data()  # [1.234, 0.00]
```

Safrone, 2018-10-10:
That works

closed 2018-10-11 by ChristianTremblay