ChristianTremblay/BAC0 #87: inactiveText Error

eloneill, 2018-05-23:
Running BAC0 version 0.99.921

Running the command:
  mycontroller = BAC0.device('2401:7', 240007, bacnet)
Yields:
  UnknownPropertyError: Unknown property 2401:7 binaryValue 4000000 inactiveText 

[Wireshark trace](https://i.imgur.com/4O04dMx.png)

ChristianTremblay, 2018-05-24:
Can you try this unelegant patch ?
In BAC0/core/io/Read.py

Lines 131

```
        if iocb.ioError:        # unsuccessful: error/reject/abort
            apdu = iocb.ioError
            reason = find_reason(apdu)
            if reason == 'segmentationNotSupported':
                self._log.warning(
                    "Segmentation not supported... will read properties one by one...")
                self._log.debug("The Request was : {}".format(args_split))
                value = self._split_the_read_request(args, arr_index)
                return value
            else:
                if reason == 'unknownProperty':
                    if 'priorityArray' in args:
                        self._log.debug('Unknown property {}'.format(args))
                    else:
                        self._log.warning('Unknown property {}'.format(args))
                    if 'description' in args:
                        return ''
                    elif 'inactiveText' in args:
                        return 'Off'
                    elif 'activeText' in args:
                        return 'On'
                    else:
                        raise UnknownPropertyError(
                            'Unknown property {}'.format(args))
                elif reason == 'unknownObject':
                    self._log.warning('Unknown object {}'.format(args))
                    raise UnknownObjectError(
                        'Unknown object {}'.format(args))
                else:
                    # Other error... consider NoResponseFromController (65)
                    # even if the realy reason is another one
                    raise NoResponseFromController(
                        "APDU Abort Reason : {}".format(reason))
```
I see in the standard that those are optional on a binary values... 

If it work, I'll try to make it more elegant....as the list could potentially grow

eloneill, 2018-05-24:
It does work now, although I get a lot of warnings.

When I run:
`mycontroller = BAC0.device('2401:9', 240009, bacnet)`

I get:

```
2018-05-24 08:02:43,729 - WARNING | Unknown property 2401:9 binaryValue 4000000 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000000 inactiveText 
2018-05-24 08:02:44,754 - WARNING | Unknown property 2401:9 binaryValue 4000000 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000000 activeText 
2018-05-24 08:02:46,857 - WARNING | Unknown property 2401:9 binaryValue 4000001 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000001 inactiveText 
2018-05-24 08:02:47,878 - WARNING | Unknown property 2401:9 binaryValue 4000001 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000001 activeText 
2018-05-24 08:02:48,993 - WARNING | Unknown property 2401:9 binaryValue 4000002 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000002 inactiveText 
2018-05-24 08:02:50,038 - WARNING | Unknown property 2401:9 binaryValue 4000002 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000002 activeText 
2018-05-24 08:02:51,155 - WARNING | Unknown property 2401:9 binaryValue 4000003 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000003 inactiveText 
2018-05-24 08:02:52,176 - WARNING | Unknown property 2401:9 binaryValue 4000003 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000003 activeText 
2018-05-24 08:02:54,299 - WARNING | Unknown property 2401:9 binaryValue 4000004 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000004 inactiveText 
2018-05-24 08:02:55,333 - WARNING | Unknown property 2401:9 binaryValue 4000004 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000004 activeText 
2018-05-24 08:02:56,450 - WARNING | Unknown property 2401:9 binaryValue 4000005 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000005 inactiveText 
2018-05-24 08:02:57,471 - WARNING | Unknown property 2401:9 binaryValue 4000005 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000005 activeText 
2018-05-24 08:02:58,588 - WARNING | Unknown property 2401:9 binaryValue 4000006 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000006 inactiveText 
2018-05-24 08:02:59,609 - WARNING | Unknown property 2401:9 binaryValue 4000006 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4000006 activeText 
2018-05-24 08:03:00,734 - WARNING | Unknown property 2401:9 binaryValue 4001000 inactiveText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4001000 inactiveText 
2018-05-24 08:03:01,756 - WARNING | Unknown property 2401:9 binaryValue 4001000 activeText 
WARNING:BAC0:Unknown property 2401:9 binaryValue 4001000 activeText 
2018-05-24 08:03:09,970 - WARNING | Device too slow, use single points polling if needed
WARNING:BAC0:Device too slow, use single points polling if needed
2018-05-24 08:03:09,973 - WARNING | Points will be read once...
WARNING:BAC0:Points will be read once...

```
Also, any tips on the last two warnings? It took almost 30 seconds to fully poll that controller.