ChristianTremblay/BAC0 #88: Device simulator

amirmazmi, 2018-06-01:
Is there a tutorial that can create a sample device with objects? That way I can run it on another computer in the network.

I have no access to a BACnet device/controller and have been trying to build a script to act as a device to simulate I-am responses and provide a response to read queries but still new to programming.

Thanks.

ChristianTremblay, 2018-06-09:
I would say that you should have a look to the bacpypes library itself. You could build some kind of application simulating a device.

BAC0 is oriented as a client application that connect to devices to read and write to them. 

See https://github.com/JoelBender/bacpypes

amirmazmi, 2018-06-10:
Thank you. 
And thank you for BAC0

closed 2018-06-10 by amirmazmi