ChristianTremblay/BAC0 #90: fx.points not working

regostar, 2018-07-19:


regostar, 2018-07-19:
[image]

[image]

regostar, 2018-07-21:
>>> import BAC0
>>> bacnet=BAC0.connect(ip="192.168.174.39/24")
Starting BAC0 version 0.99.921 (Complete)
Registered as Simple BACnet/IP App
Server started : http://192.168.174.39/24:8111
 * Serving Flask app "BAC0.web.FlaskServer" (lazy loading)
>>>  * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off

>>> bacnet.read("11:2 device 11002 objectName")
''
>>> bacnet.read("11:2 device 11002 objectList")
2018-07-21 09:36:48,028 - WARNING | Segmentation not supported... will read properties one by one...
WARNING:BAC0:Segmentation not supported... will read properties one by one...
[('device', 11002), ('analogInput', 0), ('analogInput', 1), ('analogInput', 2), ('analogInput', 3), ('analogInput', 4), ('analogInput', 5), ('analogOutput', 0), ('analogOutput', 1), ('analogOutput', 2), ('analogValue', 0), ('analogValue', 1), ('analogValue', 2), ('analogValue', 3), ('analogValue', 4), ('analogValue', 5), ('analogValue', 6), ('analogValue', 7), ('analogValue', 8), ('analogValue', 9), ('analogValue', 10), ('analogValue', 11), ('analogValue', 12), ('analogValue', 13), ('analogValue', 14), ('analogValue', 15), ('analogValue', 16), ('analogValue', 17), ('analogValue', 18), ('analogValue', 19), ('analogValue', 20), ('analogValue', 21), ('analogValue', 22), ('analogValue', 23), ('analogValue', 24), ('analogValue', 25), ('analogValue', 26), ('analogValue', 27), ('analogValue', 28), ('analogValue', 29), ('analogValue', 30), ('analogValue', 31), ('analogValue', 32), ('analogValue', 33), ('analogValue', 34), ('analogValue', 35), ('analogValue', 36), ('analogValue', 37), ('analogValue', 38), ('analogValue', 39), ('analogValue', 40), ('analogValue', 41), ('analogValue', 42), ('analogValue', 43), ('analogValue', 44), ('analogValue', 45), ('analogValue', 46), ('analogValue', 47), ('analogValue', 48), ('analogValue', 49), ('analogValue', 50), ('analogValue', 51), ('analogValue', 52), ('analogValue', 53), ('analogValue', 54), ('analogValue', 55), ('analogValue', 56), ('analogValue', 57), ('analogValue', 58), ('analogValue', 59), ('analogValue', 60), ('analogValue', 61), ('analogValue', 62), ('analogValue', 63), ('analogValue', 64), ('analogValue', 65), ('analogValue', 66), ('analogValue', 67), ('analogValue', 68), ('analogValue', 69), ('analogValue', 70), ('analogValue', 71), ('analogValue', 72), ('analogValue', 73), ('analogValue', 74), ('analogValue', 75), ('analogValue', 76), ('analogValue', 77), ('analogValue', 78), ('analogValue', 79), ('analogValue', 80), ('analogValue', 81), ('analogValue', 82), ('analogValue', 83), ('analogValue', 84), ('analogValue', 100), ('analogValue', 101), ('analogValue', 102), ('analogValue', 103), ('analogValue', 104), ('analogValue', 105), ('analogValue', 106), ('analogValue', 107), ('analogValue', 108), ('analogValue', 109), ('analogValue', 110), ('analogValue', 111), ('analogValue', 112), ('binaryInput', 0), ('binaryInput', 1), ('binaryInput', 2), ('binaryInput', 3), ('binaryInput', 4), ('binaryInput', 5), ('binaryOutput', 0), ('binaryOutput', 1), ('binaryOutput', 2), ('binaryOutput', 3), ('binaryOutput', 4), ('binaryOutput', 5), ('binaryValue', 0), ('binaryValue', 1), ('binaryValue', 2), ('binaryValue', 3), ('binaryValue', 4), ('binaryValue', 5), ('binaryValue', 6), ('binaryValue', 7), ('binaryValue', 8), ('binaryValue', 9), ('binaryValue', 10), ('binaryValue', 11), ('binaryValue', 12), ('binaryValue', 13), ('binaryValue', 14), ('binaryValue', 15), ('binaryValue', 16), ('binaryValue', 17), ('binaryValue', 18), ('binaryValue', 19), ('binaryValue', 20), ('binaryValue', 21), ('binaryValue', 22), ('binaryValue', 23), ('binaryValue', 24), ('binaryValue', 25), ('binaryValue', 26), ('binaryValue', 27), ('binaryValue', 28), ('binaryValue', 29), ('binaryValue', 30), ('binaryValue', 31), ('binaryValue', 32), ('binaryValue', 33), ('binaryValue', 34), ('binaryValue', 35), ('binaryValue', 36), ('binaryValue', 37), ('binaryValue', 38), ('binaryValue', 39), ('binaryValue', 40), ('binaryValue', 41), ('binaryValue', 42), ('binaryValue', 43), ('binaryValue', 44), ('binaryValue', 45), ('binaryValue', 46), ('binaryValue', 47), ('binaryValue', 48), ('binaryValue', 49), ('binaryValue', 50), ('binaryValue', 51), ('binaryValue', 52), ('binaryValue', 53), ('binaryValue', 54), ('binaryValue', 55), ('binaryValue', 56), ('binaryValue', 57), ('binaryValue', 58), ('binaryValue', 59), ('binaryValue', 60), ('binaryValue', 61), ('binaryValue', 62), ('binaryValue', 63), ('binaryValue', 64), ('binaryValue', 65), ('binaryValue', 66), ('binaryValue', 67), ('binaryValue', 68), ('binaryValue', 69), ('binaryValue', 70), ('binaryValue', 71), ('binaryValue', 72), ('binaryValue', 73), ('binaryValue', 74), ('binaryValue', 100), ('binaryValue', 101), ('binaryValue', 102), ('binaryValue', 103), ('binaryValue', 104), ('binaryValue', 105), ('binaryValue', 106), ('binaryValue', 107), ('binaryValue', 108), ('binaryValue', 109), ('binaryValue', 110), ('binaryValue', 111), ('binaryValue', 112), ('binaryValue', 200), ('binaryValue', 201), ('binaryValue', 202), ('binaryValue', 203), ('binaryValue', 204), ('binaryValue', 205), ('binaryValue', 206), ('binaryValue', 207), ('binaryValue', 208), ('binaryValue', 209), ('binaryValue', 210), ('binaryValue', 211), ('binaryValue', 212), ('binaryValue', 213), ('binaryValue', 214), ('file', 0), ('program', 0)]
>>> op=bacnet.read("11:2 analogValue 100 presentValue")
>>> op
31.50316619873047

THE BAC0.log file is attached
[BAC0.log](https://github.com/ChristianTremblay/BAC0/files/2215913/BAC0.log)

regostar, 2018-07-23:
Sorry sir I dont know how to name my device. It would be of great help if you could tell me how

ChristianTremblay, 2018-07-23:
You could try something like : 
```
bacnet.write('11:2 device 11002 objectName WonderfulName')
```
Not sure it will work....

ChristianTremblay, 2018-07-23:
From what I understand, those devices from ExactLogic are customizable... Maybe you can put the hand on their software and write a name to the device... technically, every BACnet device should have a name.

ChristianTremblay, 2018-07-23:
Quick suppl. note : 
12.1.1.2 Object Name
Every object shall have an Object_Name property which is a character string. Object_Name properties shall be unique within the BACnet device that maintains them. The Object_Name string shall be at least one character in length and shall consist only of printable characters.

regostar, 2018-07-24:
OK the name change worked but now I have a new error :octopus: 

`bacnet.write('11:2 device 11002 objectName DeviceName')`
`bacnet.devices`
           Manufacturer Address   Device ID
Name                                       
DeviceName   ExactLogic    11:2       11002
`mycontroller = BAC0.device('11:2', 11002, bacnet)`
2018-07-24 13:56:09,192 - WARNING | Segmentation not supported... will read properties one by one...
WARNING:BAC0:Segmentation not supported... will read properties one by one...

2018-07-24 13:56:53,803 - WARNING | Unknown property 11:2 binaryInput 0 inactiveText 
WARNING:BAC0:Unknown property 11:2 binaryInput 0 inactiveText 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 156, in __init__
    self.new_state(DeviceDisconnected)
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 166, in new_state
    self._init_state()
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 543, in _init_state
    self.connect()
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 570, in connect
    self.new_state(RPDeviceConnected)
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 166, in new_state
    self._init_state()
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 325, in _init_state
    self._buildPointList()
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/Device.py", line 390, in _buildPointList
    self.custom_object_list)
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/mixins/read_mixin.py", line 513, in _discoverPoints
    (self.read_single('{} {} inactiveText '.format(point_type, point_address))),
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/devices/mixins/read_mixin.py", line 439, in read_single
    return self.properties.network.read(request)
  File "/home/pi/.local/lib/python3.5/site-packages/BAC0/core/io/Read.py", line 147, in read
    'Unknown property {}'.format(args))
BAC0.core.io.IOExceptions.UnknownPropertyError: Unknown property 11:2 binaryInput 0 inactiveText 
`mycontroller`
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'mycontroller' is not defined

ChristianTremblay, 2018-07-24:
Are you trying the script in the REPL or is this a script (.py file) that you call ?
It's like if you try to use the variable `mycontroller` before it's ready.

Try with the command line at first just to be sure everything works. (or Jupyter)

regostar, 2018-07-27:
I am trying the scripts on the terminal (ubuntu)

ChristianTremblay, 2018-09-09:
@regostar  I know it's been a long time but can you try the develop branch. There is a new feature to use On and Off as inactiveText and activeText if they are not provided.

regostar, 2018-09-12:
Thank you will try

ChristianTremblay, 2018-09-16:
By now I've published the develop branch to master so you can also just try updating BAC0 from Pypi

closed 2018-11-22 by regostar