ChristianTremblay/BAC0 #91: Can variables be added to the emulated bacnet device?

fgervais, 2018-07-19:
I'm not really sure how to explain this but I'll give my best shot.

When we do this:

```
import BAC0
bacnet = BAC0.connect()
```
as far as I can tell, it creates a virtual bacnet device on the PC.

So from here I somehow though that we could do this:

```
bacnet['Analog Value 1'] = 44
```

so other bacnet devices can read that value from the PC through bacnet.

However I tried to find a way to do it today and I have a feeling I jumped to conclusion too quickly and that this might not be possible.

I'd just want to quickly confirm that this is or isn't possible.

regostar, 2018-07-20:
Bacnet=BAC0.connect(ip="192.168.3.3")
(IP address of the ethernet port connected to a Bacnet device)
This means that you are establishing a connection through a port from your script to the Bacnet. A virtual Bacnet device is not created.

Then
I have never seen or tried 
Bacnet['Analog Value 1']=44
I don't think it will work.
The way to do that is 
Bacnet.write('2:5 device 5 AnalogOutput 100')
This must be possible.
The reason that the former isn't possible is simple, 
Within the Bacnet.write() function,
There exists a split command , split command is used to extract the values from a string into separate variables. But that is not happening in Bacnet[]. So it won't work.But I will still give it a try and let you know.

ChristianTremblay, 2018-07-20:
@fgervais this is possible but not implemented. When creating an instance of BAC0, you are in fact defining a BACnet device with a "local device" object. This object is built with a device ID, a vendorName, etc... and could be supplied with variables at the time of creation.

If you want to play with that, have a look to `BAC0\scripts\Base.py` and `BAC0\scripts\Lite.py`.
Then look at something like : https://github.com/JoelBender/bacpypes/blob/master/samples/MultiStateValueObject.py
It shows the creation of a MultistateValueObject : 
```
 # make a multistate value object
    msvo = MultiStateValueObject(
        objectIdentifier=('multiStateValue', 1),
        objectName='My Special Object',
        presentValue=1,
        numberOfStates=3,
        stateText=ArrayOf(CharacterString)(['red', 'green', 'blue']),
        )
    _log.debug("    - msvo: %r", msvo)

    # add it to the device
    this_application.add_object(msvo)
```
As you see, you create the object you want...then you add it to the application.

In BAC0, it is referenced to `self.this_application`

Future access to this object will have to be defined (like if you want to access it by the square bracket shortcut...) and I never been so far with BAC0.... as I always use it as a client software, without local variables.

Un peu d'huile de coude.... and you should be able to reach your goal. Joel made some improvements to bacpypes so when you create an object, default parameters will be included. Look into `bacpypes\local folder`

Hope it helps

ChristianTremblay, 2018-07-20:
[suppl detail]
I would start subclassing Lite class to play with that.

fgervais, 2018-07-20:
Yes it helps a lot, I'm really excited to know that this is possible.

I'll take a look and see if can I come up with something.

Thanks

ChristianTremblay, 2018-07-23:
I'll close this and if you ever need something else, just open a new specific issue.

closed 2018-07-23 by ChristianTremblay

ChristianTremblay, 2018-10-01:
@fgervais @regostar Have a look to that : 
https://github.com/ChristianTremblay/BAC0/blob/master/tests/test_WithFakeDevice.py

This may give you hints on how I see adding devices to an instance of BAC0