ChristianTremblay/BAC0 #93: Timeout on bacnet network calls.

Safrone, 2018-07-20:
I am capturing data from a bacnet enabled device using code functionally similar to this:

```
bacnet = BAC0.connect()

device_address = ...
points = [10, 11, 12, 13, 14, 15]
requests = ["analogValue %s presentValue" for p in points]
    
def get_data(network, device_address, requests):
    return bacnet.readMultiple('%s %s' % (device_address, ' '.join(requests)))

def save_data(data):
    ...

while True:
    data = get_data(bacnet, device_address, requests)
    save_data(data)
```

This works great but every once and awhile `bacnet.readMultiple` will hang forever (I suspect an issue on the target device possibly caused by too many requests from my device). Is there any way to bake a timeout into the request to BAC0 e.g. `BAC0.connect(timeout=60)`?

I would set up my own timeout using signals but I happen to be running the get_data like code in a thread already so it would require rearchitecting my code.

ChristianTremblay, 2018-07-23:
Hummm.... 
Maybe something like : 
```
my_obj_list = [('analogValue', 10),
             ('analogValue', 11),
             ('analogValue', 12),
             ('analogValue', 13),
             ('analogValue', 14),
             ('analogValue', 15)]

dev = BAC0.device('2:5',5,bacnet, object_list = my_obj_list, poll=10)

This will poll the 6 points every 10 seconds and save the value in the memory.
You can also use the auto_save feature
https://bac0.readthedocs.io/en/latest/BAC0.core.devices.html?highlight=auto_save

if you want that script to run for really long period of time. It will save the value to SQLite locally. Other users had success using this.

You could also choose : `poll=0` and make the reading yourself...

All in memory histories are stored using the Pandas Series class if available (or simple list if Pandas is not installed).

Safrone, 2018-07-23:
My issue is that occasionally, my bacnet request will hang forever on `bacnet.readMultiple(...)`. My other code is more for example.

As this is an I/O request I feel it could have a timeout as part of it but I'm not sure if that is an issue with bacpypes or if this library could implement it.

ChristianTremblay, 2018-07-23:
What is the frequency of the readings you use ?

readMultiple will defer to the bacpypes thread the read task and wait for a result. Timeout are handled by the bacpypes stack itself at this level.

Slowing down the process could maybe help...

Safrone, 2018-07-23:
I was able to figure something out by slightly altering the readMultiple command: (`...` indicates unchanged code from current develop branch `readMultiple`)

```
def readMultipleTimeout(self, args, timeout=None):
    ...
    try:
        # build an ReadPropertyMultiple request
        iocb = IOCB(self.build_rpm_request(args))
        if timeout:
            iocb.set_timeout(timeout)  # added
        # pass to the BACnet stack
        deferred(self.this_application.request_io, iocb)
        self._log.debug("{:<20} {!r}".format('iocb', iocb))
    except ReadPropertyMultipleException as error:
        # construction error
        self._log.exception("exception: {!r}".format(error))
    
    iocb.wait()  # Wait for BACnet response
    ...
    
    if iocb.ioError:  # unsuccessful: error/reject/abort
        if isinstance(iocb.ioError, Exception):
            raise Exception("Failed to get data: {!r}".format(iocb.ioError) from apdu  # added
        apdu = iocb.ioError
        reason = find_reason(apdu)
        ...
     
```

I agree with your assessment that I could make fewer calls but this is more about robustness since I don't want my code to hang.

ChristianTremblay, 2018-07-24:
Ho... I see. I'll validate but it seems to be a good idea.

ChristianTremblay, 2018-09-29:
I talked with Joel (bacpypes) and he not sure it’s the best way to handle this kind of behaviour. I’ll keep this open but still thinking about it.

max13fr, 2018-10-11:
Hello,

I have the same issue with a script dumping all points of a device : 

```
bacnet = BAC0.connect(ip=args.localip)
device = BAC0.device(args.deviceip, args.deviceid, bacnet, poll=0)
for pt in sorted(device.points, key=lambda elem: (elem['type'], int(elem['address']))):
    print("{:<20s} | {:<5s} | {:<40s} | {:>20s} | {:<60s}".format(pt['type'], str(pt['address']), pt['name'], str(pt.value), pt['description'])
bacnet.disconnect()
```

This script is called each minute with a cron. Unfortunatelly, sometime the script hangs forever and blocks all following executions (port already in use).

Thanks in advance,
Best regards, Max

ChristianTremblay, 2018-10-11:
Using BAC0.connect() is probably too heavy for your use case.
I suspect bokeh server and flask server may have problem shuting off when you close the script...

Please try with : 

```
bacnet = BAC0.lite(ip=args.localip)
device = BAC0.device(args.deviceip, args.deviceid, bacnet, poll=0)
for pt in sorted(device.points, key=lambda elem: (elem['type'], int(elem['address']))):
    print("{:<20s} | {:<5s} | {:<40s} | {:>20s} | {:<60s}".format(pt['type'], str(pt['address']), pt['name'], str(pt.value), pt['description'])
bacnet.disconnect()
```
If you keep the app running, you can also use `point.lastValue` to "not generate" traffic as the polling will update the value.

ChristianTremblay, 2018-10-11:
Ho ! And be sure to update to 0.99.943 as a few bugs fixes the poll=0 features that could also explain the "hangs" as the parameter was not correctly handled.

max13fr, 2018-10-12:
Hello,

I was in 0.99.933. I just updated to 0.99.943 but there is a little issue with the port in BAC0.lite() : 

```
2018-10-12 13:21:47,211 - INFO    | Starting BAC0 version 0.99.943 (Lite)
Traceback (most recent call last):
  File "./remote/bacnetip/bin/bacnetip_cmd.py", line 43, in <module>
    bacnet = BAC0.lite(ip=args.localip)
  File "/usr/local/lib/python3.5/dist-packages/BAC0/scripts/Lite.py", line 81, in __init__
    ip_addr = Address('{}/{}:{}'.format(ip,mask, port))
  File "/usr/local/lib/python3.5/dist-packages/bacpypes/pdu.py", line 52, in __init__
    self.decode_address(args[0])
  File "/usr/local/lib/python3.5/dist-packages/bacpypes/pdu.py", line 278, in decode_address
    raise ValueError("unrecognized format")
ValueError: unrecognized format
```

I switched to develop branch where you fixed it (https://github.com/ChristianTremblay/BAC0/commit/e95af7c1b69d101f517b8b073957ba337b253f93).

I'll keep you informed if the problem comes back with this version.

Best regards,
Max

max13fr, 2018-10-15:
Hello,

For the update, the issue still occurs (after 1 day for this test). 

Best regards,
Max

ChristianTremblay, 2018-10-15:
Let's make another issue https://github.com/ChristianTremblay/BAC0/issues/112

ChristianTremblay, 2019-08-21:
The actual develop branch corrects a bunch of communication bugs one being the issue of whois iam that were bypassing the deferred function that makes all communication calls. I suspect that this was the cause of numerous communication glitches. Other users are reporting that develop branch. I would be grateful if you could test this branch and see if it improves your experience.

Thanks

ChristianTremblay, 2019-09-09:
Timeout is now implemented in IOCB requests. I’ll close and if needed, reopen. Thanks for your patience and feedback !

closed 2019-09-09 by ChristianTremblay

Safrone, 2019-09-09:
Awesome! I had an old version pinned for my use cases but perhaps now I'll upgrade for the other networking improvements.