ChristianTremblay/BAC0 #100: Allow to run several clients instances of BAC0.connect() on the same server

max13fr, 2018-09-18:
Hello,

I would like to run several clients instances of BAC0 on the same server (to allow debuging on specific device without stopping main production poller).

Is that possible to access to the socket to allow to add REUSE_ADDR & REUSE_PORT ?
Otherwise, is that possible to select another free port ?

Thanks in advance,
Best regards, Max

ChristianTremblay, 2018-09-18:
I don’t think it’s possible because of the way bacpypes works. Each instance of BAC0 is a BACnet device and needs to always listen to the port. It doesn’t work like a standard TCP connection where we can pause and restart later. That said, you could run a VM, or use another network card.

max13fr, 2018-09-18:
Bacpypes seems to allow changing port for this case (see http://bacpypes.sourceforge.net/gettingstarted/gettingstarted002.html#changing-ini-files).

The main process would use the default port 47808 and the secondary process on 47809 for example.

ChristianTremblay, 2018-09-18:
Yes, but as far as I know, if you try to talk to a BACnet router configured on port 47808, it will not answer to your requests on port 47809...

You would need some kind of internal route...

Technically, you can start BAC0 using another port....

And here is what I get

[image]

ChristianTremblay, 2018-09-18:
Or inside the same app...
[image]

max13fr, 2018-09-18:
Indeed broadcast features are not working (like whois) because it's always sent to port 47808.
But for interrogation on specific device, it's work like a charm (eg: BAC0.device(ip, id, bacnet)).

It would be perfect if we could reuse port 47808 only for broadcasting part (as we didn't need any routing 
intelligence because it's a broadcast, so it's for everyone).

Safrone, 2018-09-18:
An (imperfect) solution is to create essentially an API with the main bacnet network process so the read() capabilities can be accessed from another program.

ChristianTremblay, 2018-09-18:
The API is probably something close to what BACnet/Secure will be...a kind of WebSocket thing....

I tried accessing devices using 2 different BACnet networks and I have not been able to do it. Probably because devices are MSTP device behind a router configured for port 47808. Your use case may be possible using BACnet/IP devices ?
[image]

closed 2018-09-29 by ChristianTremblay