ChristianTremblay/BAC0 #99: BAC0.connect() not found then BAC0.lite() not found

max13fr, 2018-09-18:
Hello,

Firstly, thanks for your great package.
There are two issues with the current __init__.py : 

`try:
    from . import core
    from . import tasks
    from .scripts.Base import Base
    from .core.devices.Device import Device as device
    from .core.devices.Device import DeviceLoad as load
    from .tasks.Poll import SimplePoll as poll
    from .tasks.Match import Match as match
    from .core.utils.notes import update_log_level as log_level
    from .infos import __version__ as version
    
    # To be able to use the complete version pandas, flask and bokeh must be installed.
    try:
        import pandas
        import bokeh
        import flask
        _COMPLETE = True
    except ImportError:
        _COMPLETE = False
    
    if _COMPLETE:
        from .scripts.Complete import Complete as connect
        from .scripts.Lite import Lite as lite
    else:
        from .scripts.Lite import Lite as connect

except ImportError:
    pass # Probably installing the app...`

### 1) All required imports error are hidden : 

**My BAC0 was not working:**

`
>>> import BAC0                                                                                                                                                                                                                               
>>> BAC0.version                                                                                                                                                                                                                              
Traceback (most recent call last):                                                                                                                                                                                                            
  File "<stdin>", line 1, in <module>                                                                
AttributeError: module 'BAC0' has no attribute 'version'                                                                                                                                                                                      
>>> BAC0.connect()                                                                                                                                                                                                                   
Traceback (most recent call last):                                                                                                                                                                                                            
  File "<stdin>", line 1, in <module>                                                                                                                                                                                                        
AttributeError: module 'BAC0' has no attribute 'connect'                                                                                                                                                                                      
>>>  `

I found the issue by displaying the ImportError in the __init__.py : my version of bacpypes was too old : 

`>>> import BAC0                                                                                                                                                                                                                               
Traceback (most recent call last):                                                                                                                                                                                                            
  File "<stdin>", line 1, in <module>                                                                                                                                                                                                         
  File "/usr/local/lib/python3.5/dist-packages/BAC0/__init__.py", line 7, in <module>                                                                                                                                                         
    from .scripts.Base import Base                                                                   
  File "/usr/local/lib/python3.5/dist-packages/BAC0/scripts/__init__.py", line 7, in <module>                                                                                                                                                
    from . import Base                                                                                                                                                                                                                        
  File "/usr/local/lib/python3.5/dist-packages/BAC0/scripts/Base.py", line 32, in <module>                                                                                                                                                    
    from bacpypes.local.device import LocalDeviceObject                                                                                                                                                                                      
ImportError: No module named 'bacpypes.local'` 

A simple _pip3 install bacpypes --upgrade_ fix the issue.
I think that the main missing imports should raise errors (or at least display messages to allow debuging).

### 2) BAC0.lite not always available

`>>> import BAC0
>>> BAC0.lite()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'BAC0' has no attribute 'lite'`

I didn't have have pandas, bokeh and flask installed so _COMPLETE is False.
In that case BAC0.lite is not defined. It would be perfect to add .scripts.Lite as connect() AND lite() to be sure that we use the lite version (regardless of the modules available).

Thanks in advance,

Best regards,
Max

max13fr, 2018-09-18:
I'm not an python expert so I don't know the best way to handle external dependencies.

I think the main issue is hidden ImportError. The main try except should be removed or at least should print the error to allow to find the source of the issue.
But I don't know the repercussions during the installation (pass # Probably installing the app...).

ChristianTremblay, 2018-09-18:
Yeah there are repercussions when installing... (that's why I had "pass".... Now I added a few prints to give information... hope it'll clarify the process.

Safrone, 2018-09-18:
there is a way to specify optional/extra requirements in setup.py: http://peak.telecommunity.com/DevCenter/setuptools#declaring-extras-optional-features-with-their-own-dependencies

with that one can do `pip install BAC0[FULL]` or something like that

ChristianTremblay, 2018-09-18:
Thanks @Safrone 
I didn't know that, it could be something like
```
setup(
    name="BAC0",
    ...
    extras_require = {
        'lite':  ["bacpypes>=0.17.5"],
        'full': ["bacpypes>=0.17.5", "pandas", "bokeh", "flask"],
    }
)
```
I'll look into it someday. My only concern is that as I'm typically working with Windows machine, I rely on conda to install pandas, bokeh, etc... `pip install pandas` on Windows is not super easy.... 
And on Raspberry Pi, I usually use berryconda to because again, building pandas on a Pi is not super easy :-)

ChristianTremblay, 2018-09-18:
I suspect a wrong bacpypes version. Can you `pip uninstall bacpypes` then `pip install BAC0` ?

hobo0ps, 2018-09-18:
yeah that's what it was, thanks for the help, it's working now.
i have one more question: 
i'm connected to a controller and i want to look for points but nothing comes up:
[image]
It seems to be connecting to the controller but it's not reading the points.
do you know what could be causing this?

closed 2018-09-21 by ChristianTremblay