ChristianTremblay/BAC0 #104: long term trending on pi

bbartling, 2018-09-26:
Hello,

I am attempting to use BAC0 on pi for some long term trending with trending intervals of 15 minutes and a save to SLQlite option. First of all I am scratching my head on how to create a .py file and run a script as I dont think IPython/Jupyter notebooks is an option on raspberry pi.

I have a post here on raspberry stack exchange for using a crontab service to run a .py file every 15 (or maybe even 5) minutes but I am having some issues. It works just fine in terminal running the .py file but I cant get it to work with a crontab service.
https://raspberrypi.stackexchange.com/questions/89228/crontab-schedule-py-file?noredirect=1#comment142588_89228

If I scratch the crontab service idea and just try to do a loop program to save the SQLite db, it appears that the bacnet device is still getting polled at the default 10 second interval even if I define `poll=300`. Just out of curious if the bacnet device polling is every 10 seconds can the pi B+ run out of memory? Does it only save data to the microUSB card during the `.save()` else it holds the data in the computer memory?

[image]

Any tips are greatly appreciated for a best practice!
```
import BAC0
import pandas as pd
import time
import datetime

bacnet = BAC0.connect(ip='10.30.4.104/16')
hws = BAC0.device('100:99', 30099, bacnet, poll=300)

def terminate():
	controller.save()
	time.sleep(2)
	bacnet.disconnect()
	print("Its over..." , datetime.datetime.now())

for i in range(5):
	hws.save()
	print("DB saved: " , datetime.datetime.now())
	time.sleep(900)

terminate()
```

I also notice that if I define the SQLite db name, I get an error as well on pi with a similar script using something like `hws.save(db='test')`. Eventually I would like to create a script that could poll multiple BACnet devices but save to the same db.

Thank you so much for creating a such a cool tool!
Cheers,
Ben

bbartling, 2018-09-28:
One update is attempting to run the crontab  service again with executing this .py file every 15 minutes.

```
import BAC0

bacnet = BAC0.connect(ip='10.30.4.104/16')
hws = BAC0.device('100:99', 30099, bacnet, poll=300)
hws.save()

bacnet.disconnect()
```

But I get an error that 'sql' is not defined.
[image]

And if I give a db name with `hws.save(db='testnew')`, I get a keyword error:
[image]

Another odd thing is if I run the same exact script in python 3.6 (berry Conda) a BAC0 attribute error happens... Thanks any tips help if I am doing something goofy on my end.
[image]

Here is some screenshots of the python versions for 3.5 & 3.6
[image]
[image]

[image]

ChristianTremblay, 2018-10-15:
@bbartling  What if ? 
```
import BAC0

bacnet = BAC0.connect(ip='10.30.4.104/16')
hws = BAC0.device('100:99', 30099, bacnet, poll=0)
hws.save()

bacnet.disconnect()
```

Polling is not necessary as the script will run every 15 minutes....

(be sure to use an updated version of BAC0.... recently fixed a bug with poll argument.)

bbartling, 2018-10-16:
Hi Christian,

I did a pip uninstall BAC0 then pip install BAC0 on pi but it seemed to install the same exact version. I removed the `pole` like you mentioned as a Crontab service will just execute the file every 15 minutes... Here is some screenshots when I run the script in 3.5 & 3.6. Thanks any tips help! Is there a developer version I could try if I git clone BAC0? No rush on this either thanks...

[image]

ChristianTremblay, 2018-10-16:
You should be at 0.99.943.... you may have to repeat pip uninstall BAC0 a few times....until there nothing left

ChristianTremblay, 2018-10-16:
You may also check if sqlite3 is installed

bbartling, 2018-10-16:
Hi Christian,

I did install slqlite 3 as recommended:
[image]

and I did uninstall a few times then reinstall BAC0 to revision 99.943 but I still have the same issues:
[image]

ChristianTremblay, 2018-10-16:
Yeah but look into your corn Job....version 934
Can you send me the cron job ?

ChristianTremblay, 2018-10-16:
!!! Hoops.... I see what’s wrong with 3.6...
I’ve broken something in the address handling.... give me a few minutes

ChristianTremblay, 2018-10-16:
@bbartling just published a hot fix on Pypi.... 944

Safrone, 2018-10-16:
also, use `pip3.5` or `pip3.6 install BAC0` as the regular `pip` will go to your default python (which could be 2.7). You can see the BAC0 version is still 0.99.934 in the latest post.

bbartling, 2018-10-18:
Just for fun I have a new Windows 10 laptop and I installed the latest version of anaconda on it which is 3.7... And I also did a `!pip install BAC0`

Im still having issues trying to get anything to work. It appears to create an SQLite db without a name and there is no data in it when I attempt to browse with SQLite db browser.

[image]

If I am on a full blown computer and attempt to use the full version of BAC0... In the screen shot above why is it showing Lite version?
[image]

Thanks for your time & help

ChristianTremblay, 2018-10-18:
Regarding "Complete" vs "Lite"
You probably are missing some modules... 
To run Complete you need  :  
* Bokeh
* Pandas
* Flask
* flask_bootstrap

bbartling, 2018-10-22:
Hi Christian, I needed flask_bootstrap to get the full version working..

Am I doing anything wrong that BAC0 is disconnected? Is it not finding the device? If the code is Ok, I wonder if there is something wrong on the Niagara side that it isn't connecting... In Niagara the controller is online and points look Ok.

[image]

[image]

[image]

ChristianTremblay, 2018-10-22:
Router enabled on Jace ? 
IP port enabled ?
Same subnet ?

bbartling, 2018-10-24:
Hi Christian, Yes router is enabled on JACE as well as IP port...
[image]

Boiler controller is online thru an MSTP trunk on JACE n4:
[image]

IP configurations of the JACE:
[image]

IP configuration for my laptop that is also running BAC0 & workbench simultaneously. I do notice that there is two gateway addresses on my laptop IP config. But workbench seems to work just fine connecting to the JACE. -Also I think someone may have switched the Jace station to SSL recently too
[image]

I also rebooted the JACE thru platform just to try something and it didn't seem to help... Im also on a new windows 10 laptop trying this Vs pi where the most recent version of anaconda is 3.7... Any tips help thanks for your time..

Not sure why it shows disconnected... But I maybe able to switch gears and attempt to us pyhaystack for what I would like to try until my computing skills are up to par...
[image]

ChristianTremblay, 2019-09-09:
The new version include a timeout on requests, This will help in long term tasks. Please try and give me feedback. Thanks

bbartling, 2019-09-09:
👍

closed 2019-09-09 by ChristianTremblay