ChristianTremblay/BAC0 #107: BAC0 performance inquiry

tapennings, 2018-10-01:
I'm a happy user of BAC0. We use BAC0 as an interface between a Python test script on a PC and a process controller in a water disinfecting unit. In this test application, a fixed number of points is read and written to at a fixed interval, therefore a cyclic Read/ Write. I am curious about the performance of BAC0 in such an application.

**Application details**

**Infrastructure:**
- Connection over IP network from a BACnet Workstation (PC) to a BACnet-enabled process controller.

**Cyclic actions:**
1. Read the values of a number of points in this format:
mycontroller['point_name']

2. Write the values of a number of points in this format:
mycontroller['point_name'] = 23

As observed during a Wireshark BACnet analysis on my system: measured time between a single read/ write request from workstation and acknowledge of this read/ write from the controller = 0.12 [s]

Therefore, the 'net expected time' for:
Subsequent read of 80 points: 80 * 0,12 s = 9,6 [s]
Subsequent write 13 points: 13 * 0.12 = 1,6 [s]

Total net time read-write = 11 [s].

Plus, as can be expected, a certain amount of processing time in BAC0 and underlying BACpypes.

As observed in my system from an Eclipse programming environment, reading 80 points and writing 13 points over BAC0 takes 29 + 12 = 41 [s] for a single read-write cycle.

**Questions:**
1. Would this mean that the processing in BAC0 and BACpypes takes 41 - 11 = 30 [s]?
2. Is there a way to speed up the BAC0 read/ write actions by using a switch option or something in the way of BAC0.lite()?

Safrone, 2018-10-01:
The best way I found to speed up regular readings (up to 2600+ points) is to batch the readings with a `readMultiple` from the underlying network object. This takes generating a list of the point addresses you want to read and piping the vales to the right places. Obviously your device must support `readMultiple` but I believe most do. If it does not support segmentation (as mine do not), you may need to split your query. I've found you can read 60+ `presentValue` points at once generally.

I'm not sure if `writeMultiple` is implemented on BAC0, bacpypes or your device so can't comment there. 

Tldr: I avoid using the Device interface (as well as pandas/bokeh) for points I'm reading long term.

ChristianTremblay, 2018-10-01:
In my own usage, I never had to look at this with as much precision... 
It really depends I guess on your goals. What is the controller ? Are the 80 points all pertinent ? Because you could create a custom object list and iterate only the points you need.

https://bac0.readthedocs.io/en/latest/controller.html?highlight=custom#object-list

@safrone bring something with WriteMultiple. It could be implemented (it is not actually)

And yes, it is always possible to rely on "direct" bacpypes syntax and make all your calls to bacnet.readMultiple. I mean, if you don't need all the bells and whistle of the device object... 

Device not supporting segmentation would also need some attention in my implementation. I quickly turned everything to single read to "make it work"....but it could be improved.

Safrone, 2018-10-01:
for reference, I made an automatic `readMultiple` backoff:

```
import logging
logger = logging.getLogger(__name__)
from itertools import chain

class BacnetExampleClient:
  def __init__(self, network, address, object_list, max_request_size=60):
    """
    Args:
      network: BAC0.connect()
      address (str): ip address of the bacnet device desired
      object_list (): List[Tuple[object_type, object_address]]
                      e.g. [('analogInput', 1), ('analogInput', 3), ...]
      max_request_size (int): max number of requests to make at once. Degrades
                              if encounters segmentation issues.
    """
  def get_data(self):
    return self.object_list, self._bacnet_get_data()

  def _bacnet_get_data(self):
    try:
      return list(chain.from_iterable(
          self.network.readMultiple(request, timeout=self.timeout) for request in self.requests)
      )
    except SegmentationNotSupported:
      logger.warning('Segmentation error occurred with request_size = %s, reducing size => %s',
                     self.max_request_size, self.max_request_size // 2)
      if self.max_request_size <= 1:
        raise
      self.max_request_size = self.max_request_size // 2
      self._build_requests()
      return self._bacnet_get_data()

  def _build_requests(self):
    objects = ["%s %s presentValue" % (object_type, object_address)
                    for object_type, object_address in self.object_list]
    requests = ["%s %s" % (self.address, " ".join(object_batch))
                for object_batch in batch(objects, self.max_request_size)]
    self.requests = requests
```

ChristianTremblay, 2018-10-01:
Be sure also not duplicating reading... If you define a controller with default parameters, every variables will get read once every 10 seconds.... so you can use historical values to make some decisions ithout the need to read the values again.

‘ ‘ ‘ 
# Create a device and read everything once every 120seconds
Device = BAC0.device(ip,Boid,bacnet, poll=120)

# Access last value without forcing a read
Device[point].lastValue
‘ ‘ ‘

This led me to find a bug in the implementation which I hope gets totally fixed in develop branch now. Using .lastValue will not trig a network read....but dev[point] will (as we can only presume no reading has been made for some time....even if it’s not the case... lastValue is pretty clear (in my head at least) that I don’t want another reading on the network...) 

PS : Please note I fixed a bug where poll param was not correctly handled... 0.99.941 contains the fix (or the develop branch if you are using it)

ChristianTremblay, 2018-10-01:
Forgot....
You can also pass a list to device
```
lst = [‘ZN1-T’, ‘ZN2-T’,’ZN3-T’]
Device[lst]
# it will return a data frame containing histories of the points
# you could use something like that to get the last value of all reading
Device[lst].fillna(method=‘ffill’)[-1:]
```

ChristianTremblay, 2018-10-01:
@Safrone may I use that inside BAC0 ? Not sure yet where (under bacnet...or the device...) but it could be very handy in certain situations to have this.

Safrone, 2018-10-03:
@ChristianTremblay Feel free!

tapennings, 2018-10-04:
Hello Christian and Safrone. My first action was to upgrade my machine to the current BAC0 version 0.99.943. The next step was making use of the polling parameter setting in BAC0.device, this was set to a required update rate for 'responsive interaction' with the water disinfecting unit. The last action was to use the proposed 'controller.points[index].lastValue' format in cyclic point value updates.

This did the job, it has resulted in a considerable update performance improvement! A cyclic readMultiple request is generated in the background at the set polling rate, as could be observed during a Wireshark-BACnet analysis of network traffic between PC and controller. From a Python application viewpoint, requesting 80 points now takes something like 0.8 [s] instead of the previous 30 [s]. As writeMultiple is not (yet) possible in BAC0, the total write time stayed the same at 10 [s].

So thank you both for your support!

closed 2018-10-04 by tapennings

ChristianTremblay, 2018-10-04:
Nice ! I’ll try to see what I can do for write multiple... I could need a few weeks :) But with the script @Safrone  sent... I’m pretty sure I can do something useful.
Thanks for letting me know it works for you. I really appreciate that kind of feedback.