ChristianTremblay/BAC0 #112: bacnet.disconnect() sometimes fails to close socket

ChristianTremblay, 2018-10-15:
When used in a script, sometimes, using bacnet.disconnect() is not sufficient. The port stays open and it's impossible to connect again.

@max13fr

ChristianTremblay, 2018-10-15:
Original script is : 
```
bacnet = BAC0.connect(ip=args.localip)
device = BAC0.device(args.deviceip, args.deviceid, bacnet, poll=0)
for pt in sorted(device.points, key=lambda elem: (elem['type'], int(elem['address']))):
    print("{:<20s} | {:<5s} | {:<40s} | {:>20s} | {:<60s}".format(pt['type'], str(pt['address']), pt['name'], str(pt.value), pt['description'])
bacnet.disconnect()
```

I did test it on my side and what I see is that the call for `pt.value` could be a source of trouble. In fact, this call will generate a read action on the network for each of the points... but technically, as the device have just been discovered, the value have already been read.

A read on each point will slow the process and can, in certain circumstances take more time than evaluated. In this case, the script runs evry minute. It is not impossible than the single read of every points take more than 1 minutes. The Cron job may try to start another job when the socket is not released yet.

I suggest trying this : 
```
bacnet = BAC0.connect(ip=args.localip)
device = BAC0.device(args.deviceip, args.deviceid, bacnet, poll=0)
for pt in sorted(device.points, key=lambda elem: (elem['type'], int(elem['address']))):
    print("{:<20s} | {:<5s} | {:<40s} | {:>20s} | {:<60s}".format(pt['type'], str(pt['address']), pt['name'], str(pt.lastValue), pt['description'])
bacnet.disconnect()
```
Replacing pt.value byt pt.lastValue will greatly accelerate the script and minimize the risk of it taking more than 1 minute to run.

max13fr, 2018-10-16:
Hello,

Thanks for the help, I try this fix and I'll keep you update.

Best regards,
Max

max13fr, 2018-10-17:
Hi, the issue still occurs even with lastValue.

Here my current code : 

```
bacnet = BAC0.lite(ip=args.localip)
device = BAC0.device(args.deviceip, args.deviceid, bacnet, poll=0)
for pt in sorted(device.points, key=lambda elem: (elem['type'], int(elem['address']))):
    print("{:<20s} | {:<5s} | {:<40s} | {:>20s} | {:<60s}".format(pt['type'], str(pt['address']), pt['name'], str(pt.lastValue), pt['description'])
bacnet.disconnect()
```

I think the issue is not coming from disconnect() but from **device.points** : the script is stuck and never stop. Perhaps we can add a timeout parameter to BAC0.device() ?

Thanks in advance,
Best regards, Max

ChristianTremblay, 2018-10-18:
what's in your ~\.BAC0\BAC0.log ?

max13fr, 2018-10-19:
Hi,

I just cleaned my BAC0.log file and disabled my temporary cron (killall --older-than 10m bacnetip_cmd.py).

I'll put the log here on the next freeze.

Best regards, Max

max13fr, 2018-10-22:
Hi,

The script is stuck but I didn't have anything in ~/.BAC0/BAC0.log

Max

ChristianTremblay, 2018-10-22:
Nothing nothing ? or nothing relevant ?

max13fr, 2018-10-23:
I'm currently trying two things : 
- add a device.disconnect() before bacnet.disconnect() in addition
- call specifics points list instead of everything

I'll keep you update

max13fr, 2018-10-23:
Unfortunatelly, even with the 2 additions above the problem still occurs.

max13fr, 2018-10-23:
Currently, I fixed the issue with a timeout decorator : 

```
import BAC0, timeout_decorator

@timeout_decorator.timeout(15) # fix infinite lock in BAC0 
def device_poll(bacnet, data, objects_list):
        device = BAC0.device(data['global_params']['device_ip'], data['global_params']['device_id'], bacnet, poll=0, object_list = objects_list)
        device.disconnect()
        return device
```

max13fr, 2018-10-25:
This fix still doesn't works : as BAC0 start threads the timeout_decorator is not working (it uses SIGALARM that only works on main thread).

It would be perfect to have a timeout argument to BAC0.device() (default to 0 to avoid issues with current scripts).

Best regards,
Max

ChristianTremblay, 2018-10-25:
And what if you would not use the device and simply use bacnet.read ? If you don’t need all the options... you’ll use only bacpypes code.

```
# Let’s pretend the device is 2:5
import BAC0
bacnet = BAC0.lite()
lst = [(“analogValue”,1), (“analogValue”,2)]
for each in lst:
    print(bacnet.read( “2:5 {} {} presentValue”).format(each[0],each[1]))
   # or
    print(bacnet.readMultiple(“2:5 {} {} presentValue units.format(each[0],each[1]))
bacnet.disconnect()
```

max13fr, 2018-10-25:
In fact, it could be a really nice idea. However, my devices are on IP how can I specify them ?

ChristianTremblay, 2018-10-25:
```
# Let’s pretend the device is 192.168.10.23
import BAC0
bacnet = BAC0.lite()
lst = [(“analogValue”,1), (“analogValue”,2)]
for each in lst:
    print(bacnet.read( “192.168.10.23 {} {} presentValue”).format(each[0],each[1]))
   # or
    print(bacnet.readMultiple(“192.168.10.23 {} {} presentValue units.format(each[0],each[1]))
bacnet.disconnect()
```

ChristianTremblay, 2018-10-25:
Also, you may want to see this : https://github.com/ChristianTremblay/BAC0/issues/107#issuecomment-425974690

It contains an interesting way of building readMultiple request... I didn't wrap my head around yet... but it may help you.

max13fr, 2018-10-26:
It's perfect, thanks for your help !

I'll keep you update.

ChristianTremblay, 2019-09-09:
@max13fr 
New version (19.9.8) now includes a timeout in IOCB request and I fixed really annoying bugs when disconnecting devices and network. Hope you'll like the new features.

closed 2019-09-09 by ChristianTremblay