ChristianTremblay/BAC0 #114: Connection to specific network adapter fails in recent BAC0 version

tapennings, 2018-10-18:
We use the following command in an existing and previously correctly functioning Python application:

BAC0.connect(ip='192.168.1.22/24')

This syntax failed with recent version 0.99.943 of BAC0. The error following upon this command '...not enough arguments, 1 instead of expected 2 (../..)...'  ( / = probably IP address and subnet mask separator?) suggests that BAC0 is no longer able to decompose an IP address & subnet mask string in CIDR notation into separate elements 'device IP address' and 'subnet mask'. 

My workaround for this issue was to downgrade both BAC0 (to 0.99.921) and BACpypes (to 0.17.0). The mentioned BAC0.connect(…) then functioned correctly again, I could establish a communication link to the specific network on my PC on which the BACnet devices reside.

ChristianTremblay, 2018-10-18:
Sorry for that.... I've broken something recently and it's been fixed in https://github.com/ChristianTremblay/BAC0/releases/tag/v0.99.944

Please let me know if everything is ok for you when updated.

I recently started to make more tests as I found a way to test BAC0 with TravisCI. It is not perfect and as I'm using the real socket while testing instead of "mocking" them, it sometimes create weird issues. But anyway. I created local test for ip, mask and port handling and they tell me it's working now.

Root cause is that I created and "port" argument when creating an instance that allow me to use
```
bacnet = BAC0.connect(port=47809)
```
This way I can define 2 instances that can talk to each other and test read, write, etc.... 
Still a work in progress but I'm pretty sure it'll make the soft stronger.

Again, sorry for the trouble.

tapennings, 2018-10-19:
Yes, everything was OK again upon updating my system to BAC0 version 0.99.944.  'Merci beaucoup' for the speedy fix and reply!

closed 2018-10-19 by tapennings