ChristianTremblay/BAC0 #117: BAC0 crash upon pandas read-write error

tapennings, 2018-11-09:
We use BAC0 as an interface between a Python test script application on a PC and a BACnet-capable process controller in a water unit. The following error occurred while running the test script application:

[image]

The syntax **device.points[index].lastValue** is used to retrieve current BACnet object present values. The poll parameter in BAC0.device(... poll = ...,) has been set to 15 seconds. It looks like pandas attempts to read or write particular values in a read-write buffer (list), and that an index error occurs.

The error happens from time to time, for instance once in every 3 test sessions with our Python test script. It takes between 5 to 20 minutes from starting the application before the crash occurs. 

Questions:
- Has this error been reported before?
- There is a dependency between BAC0 and pandas. Is there some kind of code functionality mismatch between recent versions of BAC0 and pandas?

tapennings, 2018-11-09:
The versions I have used initially are:
BAC0: 0.99.944
pandas: 0.23.4

In an attempt to work around the issue, I have downgraded the PC to BAC0 at 0.99.921 and pandas at 0.22.0. This did not solve the problem; the error re-occurred. The screen grab in my issue entry above is from the downgraded state.

Yes, I have had a look at pandas. It looks very suitable for all kinds a data set operations.

ChristianTremblay, 2018-11-09:
When this happen, would it be possible to validate : 
`dev['point']._history.value` and `dev['point']._history.timestamp`

When point are read, I only append the new data/timestamp to those 2 lists.
When someone ask for history, I create the pandas Series on-demand.
I guess I will have to provide a try/except block more robust to handle edge cases where there is a difference in length between the two lists...  
Or be pro-active and make more validation when we write to the lists...

It would help if I could get what's in those list... why they are different.

tapennings, 2018-11-12:
I have added try-except- debug logger code to our BACnet I/O application to capture those list contents at the ValueError event. As reproduction of the error is not so easy, it might take some time before I can report the results back to you.

ChristianTremblay, 2018-11-12:
I should probably look into changing the way I created the `_history` variable.
It is actually a namedTuple containing lists.
I should maybe switch that for a list of namedTuple.

Would not be too hard and it probably fix those kind of weird bugs that are maybe caused by a thread being called just between the two calls for writing to timestamp and value.

tapennings, 2018-11-20:
In an attempt to reproduce the value error, I have performed 5 complete test runs with the test script on an operational water disinfecting unit (system under test). Unfortunately (or luckily...) the error did not occur anymore.  We will therefore monitor actual use of the application by our production department to see whether the crash re-occurs. The required specific timestamp - value list contents will then be available in a debug logger file. I will report them back to you.

ChristianTremblay, 2019-06-20:
Closing, if the bug ever reappeared, please re-open

closed 2019-06-20 by ChristianTremblay

tapennings, 2019-06-21:
OK Christian, the bug has not re-occurred over the past 8 months, fine.