ChristianTremblay/BAC0 #119: ERROR:BAC0:an error has occurred: [Errno 49] Can't assign requested address

pisuke, 2018-11-21:
Hi, I have this simple program:

```
#!/usr/bin/env python3
import BAC0
import netifaces as ni
from netaddr import IPAddress
interface = 'eth0'
ni.ifaddresses(interface)
ipa = ni.ifaddresses(interface)[ni.AF_INET][0]['addr']
nm4 = ni.ifaddresses(interface)[ni.AF_INET][0]['netmask']
nm = IPAddress(nm4).netmask_bits()
ip_addr = "%s/%s" % (ipa,nm)
bacnet = BAC0.lite(ip_addr)
```
I get the following error unfortunately:
10.20.52.121/8
2018-11-21 23:01:35,152 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2018-11-21 23:01:35,153 - INFO    | Using ip : 10.20.52.121
2018-11-21 23:01:35,175 - ERROR   | an error has occurred: [Errno 99] Cannot assign requested address
ERROR:BAC0:an error has occurred: [Errno 99] Cannot assign requested address
Traceback (most recent call last):
  File "bacnet_test.py", line 18, in <module>
    bacnet = BAC0.lite(ip_addr)
  File "/usr/local/lib/python3.4/dist-packages/BAC0-0.99.944-py3.4.egg/BAC0/scripts/Lite.py", line 96, in __init__
  File "/usr/local/lib/python3.4/dist-packages/BAC0-0.99.944-py3.4.egg/BAC0/scripts/Lite.py", line 100, in update_whois
  File "/usr/local/lib/python3.4/dist-packages/BAC0-0.99.944-py3.4.egg/BAC0/core/functions/WhoisIAm.py", line 57, in whois
BAC0.core.io.IOExceptions.ApplicationNotStarted: BACnet stack not running - use startApp()

Not sure if there is anything wrong in the script, as it's pretty basic.
Anything I should look at to troubleshoot?

Thank you!
Francesco

ChristianTremblay, 2018-11-22:
What if you specify : 
```
bacnet = BAC0.lite(ip='10.20.52.121/8')
```

ChristianTremblay, 2018-11-22:
humm.... it should not make any difference on your side...
I tried setting a mask of 8 here and it works (was thinking about the mask could be a problem... but no)

(brain storming here...)
 Is the IP active ?
Is it really the IP address in use ?
blocked by another process ? (another BAC0 instance would trigger a socket error...)

Also, found out that Errno99 is related to socket module and there is a bunch of Stack Overflow issues like this one
https://stackoverflow.com/questions/19246103/socket-errorerrno-99-cannot-assign-requested-address-and-namespace-in-python

Maybe this can help

pisuke, 2018-11-24:
Thanks Christian for looking into this.
Yes, the IP is active and the one in use:

eth0      Link encap:Ethernet  HWaddr b8:27:eb:fc:cb:c7
          inet addr:10.20.52.121  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::47a2:a6dd:d50e:f65f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:457879 errors:0 dropped:10 overruns:0 frame:0
          TX packets:28912 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:35389911 (33.7 MiB)  TX bytes:3424654 (3.2 MiB)

There isn't any other BAC0 instance.
I can ping network devices in the same network.

I also tried with wlan0 and it makes the same error.

The only network interface that doesn't output the error is lo 127.0.0.1

Anywhere else I can look at? I've tried looking at the Errno99 messages but could not get to anything useful.

Thank you!

ChristianTremblay, 2018-11-25:
Could it be related to some kind of permission ? 
Trying with sudo may give a hint ?

pisuke, 2018-11-25:
I had been running the script with sudo already.
But then I tried using bacpypes directly and was able to assign that IP.
So I tinkered a bit with the Lite.py code and managed to make it work.
It was a parsing problem.
I've just done a pull request  https://github.com/ChristianTremblay/BAC0/pull/120, hope OK.
Thanks!

ChristianTremblay, 2018-11-26:
[insert curse word here]
I broke it when I made tests to be able to generate 2 different instances with different ports...
Thanks for pointing that

closed 2018-11-27 by ChristianTremblay